TBXin/NFirmwareEditor #117: myEvic configuration support and VTC Mini - Expand 1st and 2nd line too #106

BigBadBlackWolf, 2016-12-17:
When is myEvic configuration supported and what is with the announcement from #106 "VTC Mini - Expand 1st and 2nd line too" - "will release our firmware in which you can change every line, even battery line using our configuration utility."

TBXin, 2016-12-17:
1. I dont know.
2. [ArcticFox](https://github.com/maelstrom2001/ArcticFox) firmware is released for a long time.

BigBadBlackWolf, 2016-12-17:
I know the ArcticFox has appeared, but so far this is not compatible with the myEvic firmware, since NFE Tool does not support myEvic configuration yet.

TBXin, 2016-12-17:
Ehm... ArcticFox and myEvic are two different firmwares...
If you want line customization in myEvic like we make in the ArcticFox you should ask @ClockSelect make it. 
This kind of things should be done in both: firmware and configuration utility.

BigBadBlackWolf, 2016-12-19:
Okay, Thanks!
I ask @ClockSelect

closed 2016-12-22 by TBXin

nbraunstein, 2017-01-24:
Has @ClockSelect been working with you on making this a possibility?