TBXin/NFirmwareEditor #123: Width and height import issue.

ArionWT, 2017-01-06:
Во время экспорта респака, задаются атрибуты Width и Height, однако при импорте они не учитываются.

TBXin, 2017-01-06:
Так было задумано изначально, чтобы набор ресурсов не мог повредить прошивку.

ArionWT, 2017-01-06:
А если допустить возможность уменьшения размеров?

TBXin, 2017-01-06:
Объясню ситуацию:
Наборы ресурсов это безопасный с точки зрения пользователя инструмент. Их можно применять не боясь ничего сломать.

Почему уменьшение тоже может вредить?
Допустим есть стоковый глиф 6х8. Устанавливается пак, в котором этот глиф перерисован до 4х6.
Через какое-то время пользователь захочет установить другой набор ресурсов, где этот глиф такого же размера как и стоковый 6х8, но при импорте мы размер увеличивать не можем и получится плохо.

Метаданных о том какими глифы были когда-то нигде нет, так что восстановить к заводскому состоянию можно только руками и нужно знать точные размеры глифов.

ArionWT, 2017-01-06:
Если размер глифа был изменен, к примеру с 6х8 на 4х6, а после прошивка была сохранена, то восстановление прежнего размера в 6х8 через "Resize image" приведет к появлению ошибок?
Если нет, то возможно ли хранить данные об оригинальном (или предыдущем) размере в .respack и сравнивать:  идет увеличение или уменьшение размеров?
Просто из-за этой проблемы ограничивается возможность использования ресурс паков, так как из-за разных кеглей и толщин шрифтов появляются лишние отступы между символами в тексте.

TBXin, 2017-01-06:
> Если размер глифа был изменен, к примеру с 6х8 на 4х6, а после прошивка была сохранена, то восстановление прежнего размера в 6х8 через "Resize image" приведет к появлению ошибок?

Нет. Восстановить будет можно, но это ручная операция опять же.

> Если нет, то возможно ли хранить данные об оригинальном (или предыдущем) размере в .respack и сравнивать: идет увеличение или уменьшение размеров?

Оригинальные взять неоткуда, а предыдущие тоже не показательны. Допустим опять же, была прошивка с глифами 4х6 после применения какого-то пака, и на основе неё вы сделали глифы 6х8 и сохранить пак. В предыдущих будет 4х6.

closed 2017-01-06 by ArionWT