TBXin/NFirmwareEditor #126: Wismec Rx300

GGoldRock2k, 2017-01-08:
Do you gonna add Wismec RX300 to edit the firmware, as I already got it and want to edit it.

maelstrom2001, 2017-01-09:
[RX300 5.04.zip](https://github.com/TBXin/NFirmwareEditor/files/693829/RX300.5.04.zip)
Download and unpack RX300 5.04.xml to Definitions folder.

uhfnet, 2017-01-12:
Could you please also provide definitions for La Petite box by Vaponaute if possible? Seems to be identical to the VTC Mini, considering Vaponaute has a firmware v3.03 for download.

Thanks and sorry for hijacking the thread, thought it would be better than opening a new issue!

TBXin, 2017-01-14:
@uhfnet where we can download La Petite box firmware?

uhfnet, 2017-01-14:
@TBXin you can find a download link for the stock firmware here:

https://vaponaute.com/en/aircrafts/61-la-petite-box.html

Apparently, they have decided to host the firmware on Google Drive. Not very professional, but at least the device itself seems promising...

Thanks again!

TBXin, 2017-01-14:
@uhfnet oh my god! thanks, I could not even think that download link will be made in the form of plain text ... without any indications...

maelstrom2001, 2017-01-14:
@uhfnet Extract attached file to Definitions folder:
[La_Petite_Box_V3.03.zip](https://github.com/TBXin/NFirmwareEditor/files/706233/La_Petite_Box_V3.03.zip)

Actually, it will add 2nd strings block only, automatic recognition works fine.

closed 2017-01-24 by TBXin