TBXin/NFirmwareEditor #129: Pay-per-feature

febs, 2017-01-13:
I would happily pay to get NFirmwareEditor running on Linux.
I could tip ten euros. I know it is not much fur such a porting job, but maybe if you would rise a bounty you could find more people interested.

After all this is a nerdy thing :-) and I found other users installing a VM only to use this software. Maybe they would be interested too.

Just wondering. That's it. Thanks for reading.

TBXin, 2017-01-13:
If someone decides to help the project and to port the app to other platforms - we will be happy to help, but we're not going to do it.

FrankBKK, 2017-01-14:
I am running it on Virtualbox - you just have to make sure the Virtualbox Extension Pack for USB support is installed. 
Works fine on macOS and Linux Mint

Wine would also be an option but I did not try that yet ...

febs, 2017-01-14:
Wine does not work. Virtual box does, but it's a pain in the ass on this
old laptop, it takes like about 20 minutes overall to make a change...

Anyways, that's it. :-P

nbraunstein, 2017-01-24:
What are the costs associated with creating an OS X version?

closed 2017-01-24 by TBXin

Jsl-1, 2017-01-24:
Hi

I managed to make a limited version working on linux using mono.  For now, only ArcticFox configuration is working. 

I have to make some fixes, disable not working features, and update readme before publish.
I'll tell you here as soon as it's ready to test. (one or two weeks)

febs, 2017-01-24:
Hey! Thank you so much!

I cloned the repo like a few minutes ago but I was so clueless about how to
move any further. :-(

Did you clone the repo somewhere? I'd love to beta-test it if you'd like so.

Jsl-1, 2017-01-24:
Hi, my fork is not up to date and configuration is not saving, you have to wait ;) 
Sorry, I will do my best to release a beta as soon as possible ;)

febs, 2017-01-24:
No problem :-)

Anyways github is for development, nobody expects a working release if it's
not in the "master" branch :-)

I'll just wait of course, and thanks for your work!

Jsl-1, 2017-01-24:
Hi,

You can test this build

[https://ci.appveyor.com/project/jsl1/nfirmwareeditor/build/6.4.22/artifacts]
- You needs arcticfox nightly build version 170106 minimum because in is based on NFirmwareEditor nightly build , 
- You need to install mono : [http://www.mono-project.com/docs/getting-started/install/linux/](http://www.mono-project.com/docs/getting-started/install/linux/)

- NFirmwareEditor is not working, only NToolbox
- PowerCurve editing seem to be broken, will check later
- You should not launch multiple instance of the application at same time, security is disabled
- It's not fully tested, so i am not responsible if something goes wrong with your device

You can report bugs on https://github.com/Jsl-1/NFirmwareEditor/issues
Source code is available in linux branch (will be moved to mono branch soon),

Maybe it's working on Mac, I have not tested yet because I don't own a mac ;)

I will publish instruction in a new README when i have more time.

Thanks for testing ;)

TBXin, 2017-01-24:
@Jsl-1 Hid is not working? ^^

[image]

Jsl-1, 2017-01-24:
Hi TBXin !

It's working for me on ubuntu 16.04, witch is yours ?

TBXin, 2017-01-24:
@Jsl-1 16.04

Jsl-1, 2017-01-24:
Can you replace Hidsharp dll with these dll : [https://github.com/Jsl-1/NFirmwareEditor/tree/linux/src/lib](https://github.com/Jsl-1/NFirmwareEditor/tree/linux/src/lib) ( maybe a deployment problem, i have not tested the appveyor  zipped archive :/)

Jsl-1, 2017-01-24:
@TBXIN if it's not working, can you tell me the libud and libc version ? (locate libudev.so / libc.so)

Thanks!

TBXin, 2017-01-24:
@Jsl-1 
libudev0_175-0ubuntu9.10_i386
libudev0_175-0ubuntu9.10_amd64

TBXin, 2017-01-24:
@Jsl-1 i was testing at clean ubuntu installation, just install mono.
Seems to be libudev is not included, so i've downloaded and install it manually.

Jsl-1, 2017-01-24:
@TBXIN
Ok, i don't know why, but it seems to be old libudev version, in this case HidSharp don't need to be patched, it should be working with regular HidSharp nuget package . You can try with dll from HidSharp 1.5 if you have some time. I should modify hidsharp to detect version. I have assumed that all recent distribution use libudev1 :/ It's a mistake ;) 
I will check tomorrow, maybe it could work everywhere with standard hidsharp package. an my patch is useless ;)

TBXin, 2017-01-24:
@Jsl-1 i would appreciate if you tell me where i should get libudev ^^
At this point i am reinstalling ubuntu 😄

Jsl-1, 2017-01-24:
@TBXin  
 apt-get install lubudev1 ?
I have no machine at this time, i can check the package tomorrow ;)

TBXin, 2017-01-24:
@Jsl-1 i receive something like that "Package 'libudev1' has no installation candidate"

TBXin, 2017-01-24:
@Jsl-1 got success with a new clean ubuntu installation.

1. Install Mono;
2. Install libudev1 via sudo apt-get install libudev1;
3. Replace binaries (take from tree/linux/src/lib);
4. Run NToolbox.exe with a root privilegies (HID working only with sudo);

Jsl-1, 2017-01-24:
@TBXin 
Thanks for testing ;) i'll correct wrong reference in the build as soon as possible ;)

Jsl-1, 2017-01-25:
@TBXin 
Hi! 
You can't skip this step : 3. Replace binaries (take from tree/linux/src/lib);
I have just testing the zip, dll is ok and its working

Next step for me : 
- some correction are needed on windows and control size
- find a way to make replacement for cross application synchronization (can't use mutex in mono)
- implement another method for usb in HidSharp for making it work for non root users (find a way using Usb Hid api in this repository : [https://github.com/arvydas/BlinkStickDotNet/tree/master/Components/HidSharp/Platform](https://github.com/arvydas/BlinkStickDotNet/tree/master/Components/HidSharp/Platform))
- Try to make monitor working, maybe a System.Drawing.Graphics mono implementation issue
- Cleanup build configuration

I don't think windows forms mono implementation is a long term solution for having your soft working on Linux/Mac, but I don't find a better way for now ;) 

If you are interested in creating your own mono branch, I can hep to maintain ;)

TBXin, 2017-01-25:
@Jsl-1 
Hi,
> You can't skip this step : 3. Replace binaries (take from tree/linux/src/lib);

I can't get yesterday working HID without replacing binaries.

> find a way to make replacement for cross application synchronization (can't use mutex in mono)

Not only mutex ^^ also RegisterWindowsMessage.

> Try to make monitor working, maybe a System.Drawing.Graphics mono implementation issue

I don't try to debug on linux, but i think monitoring issue are same as with any curves editors, because all of them are using Chart control.

> I don't think windows forms mono implementation is a long term solution for having your soft working on Linux/Mac, but I don't find a better way for now ;)

Yes, you are right. This is a workaround and not more. If i remember correctly there are no welldone created/designed cross-platform UI frameworks for .NET that are gives posibilities to create rich and complex UI.

So... Better way, IMHO - it's a create fully native application for each platform, which will copies original NToolbox behaviors.

TBXin, 2017-01-25:
@Jsl-1 
P.S. Did you get working firmware updater on Linux? Mine is stops after device reboot.

Jsl-1, 2017-01-25:
@TBXin 

> I can't get yesterday working HID without replacing binaries.
Ok, I'll check all reference ;)

> P.S. Did you get working firmware updater on Linux? Mine is stops after device reboot.
I Confirm, same problem ;) 

I'll tell you when a new build is created ;)

Jsl-1, 2017-01-25:
@TBXin 

>P.S. Did you get working firmware updater on Linux? Mine is stops after device reboot.

After flashing, device reboot sucessfully, but it wait for  LoadFromLDRow which always return false.
Some idea ?
There's also a multithreading issue with InfoBox.Show on a thread, so I call UpdateUI to show these box. At least, this does not crash the program after flashing.

> I can't get yesterday working HID without replacing binaries.

Found and remove a nuget package reference witch can cause overriding library in final package ;) 

Lastest build  in  [https://ci.appveyor.com/project/jsl1/nfirmwareeditor/history](https://ci.appveyor.com/project/jsl1/nfirmwareeditor/history)

Jsl-1, 2017-01-25:
@TBXin 

System.Windows.Forms.DataVisualization.Charting.Chart is not Implemented in mono :/
So I can't make Curves Editor and Monitor Working.

TBXin, 2017-01-25:
@Jsl-1 ha) for some reason I thought so.

TBXin, 2017-01-25:
> After flashing, device reboot sucessfully, but it wait for LoadFromLDRow which always return false.
Some idea ?

It's not after, it's before flashing. First we need boot to the bootloader, for this operation we need update dataflash and reboot device, after reboot we are checking dataflash for bootloader flag (LoadFromLDRom).

Device is rebooted without any problmes, but dataflash is not updating, thats why device starts normally, not in the bootloader.

Jsl-1, 2017-01-26:
@TBXin 
Hi,
If I set LoadFromLDRow = true and write dataflash,  the value is always false after reading again without rebooting device

TBXin, 2017-01-26:
@Jsl-1 it can mean only one thing, dataflash write is not working on the linux for some reason.
I can suggest that there are issue in the hid stream, rather in the linux hid wrapper.

Jsl-1, 2017-01-26:
But device configuration use the same method  HidConnector.WriteDataflash and its working...

TBXin, 2017-01-26:
@Jsl-1 nope, configurator does not use write dataflash command.

Jsl-1, 2017-01-26:
@TBXin sorry, there's some issues with my brain today :/

Jsl-1, 2017-01-26:
@TBXin  nobody should use this for now, I remove all downloads, because hidsharp loose some bytes when writing ;) Configuration upload is broken to ;)

Jsl-1, 2017-01-30:
Hi,

@febs, I managed to make it work, a new build is available.

@TBXin 
There where an issue  with usb transfert report byte being removed by hidsharp, which cause unexpected results when transfer size is greater than max output report size

I managed to update to arctifox 170129 from linux with current build.

febs, 2017-01-30:
Nice, thanks!