TBXin/NFirmwareEditor #130: [Vtwo Mini] Changing batteries, resetting stats and confirming "old coil" causes firing at 30W.

forkoz, 2017-01-16:
I'm running the 1/16 release on my vtwo mini and using a 1.4 ohm coil with SS316L. When I change batteries, reset stats and keep the old coil the mod fires at 30W instead of the 10 it was set to. I have to go into toolbox and re-upload the profile for it to stop. If I select "new" coil it gives a TCR error and will not fire.

[fire.csv.txt](https://github.com/TBXin/NFirmwareEditor/files/709352/fire.csv.txt)
[image]

maelstrom2001, 2017-01-16:
30 W is standard value for TCR Check procedure. When you changed coil/restart box/select another TC profile etc. box will check your coil on TCR. It starts from 30 to 40 W regarding selected temperature settings in profile. Your coil is not passing this check, 'cause resistance grows higher than 1,5 Ohms.

TBXin, 2017-01-16:
You should ask questions about ArcticFox here: https://github.com/maelstrom2001/ArcticFox/issues

closed 2017-01-16 by TBXin