TBXin/NFirmwareEditor #141: German Translation (Deutsche Übersetzung) for Language Pack from NToolbox

BigBadBlackWolf, 2017-01-22:
Hallo,
I have made a German Translation for NToolbox. How I can upload it here?

TBXin, 2017-01-22:
Hello,

You can upload it to any file hosting site and give me a link.

BigBadBlackWolf, 2017-01-22:
Hallo, here is my German Translation (Deutsche Übersetzung) for the
Languages Pack from NToolbox. I hope you can need it for next upgrade of
your very good software.

BigBadBlackWolf

// Shared
Toolbox.Shared.SecondString=Sekunde
Toolbox.Shared.SecondsString=Sekunden

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox Konfiguration
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic Konfiguration
Toolbox.MainWindow.DeviceMonitorButton=Geräteüberwachung
Toolbox.MainWindow.ScreenshooterButton=Displayabbildung
Toolbox.MainWindow.FirmwareUpdaterButton=Firmware-Aktualisierung
Toolbox.MainWindow.AboutLink=NFE Info && Links

Toolbox.MainWindow.Tray.ShowToolbox=Zeige NFE Toolbox Fenster
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Öffne ArcticFox Konfiguration, wenn Gerät verbunden ist
Toolbox.MainWindow.Tray.TimeSync=Synchronisiere Zeit, wenn Gerät verbunden ist
Toolbox.MainWindow.Tray.StartWithWindows=Starte NFE Toolbox, wenn Windows startet
Toolbox.MainWindow.Tray.Exit=Beenden

// ArcticFox Configuration Window
Toolbox.ArcticFoxConfiguration.Message.ConnectDevice=Verbinde Gerät mit\n\nArcticFox\n[{0}]\n\nFirmware oder neuere
Toolbox.ArcticFoxConfiguration.Message.DownloadingSettings=Einstellungen runterladen...
Toolbox.ArcticFoxConfiguration.Message.OutdatedToolbox=NFE Toolbox ist veraltet.\n\nUm fortzufahren, laden Sie bitte\n\nletzte verfügbare Version herunter.
Toolbox.ArcticFoxConfiguration.Message.NoCompatibleUSBDevice=Kein kompatibeles USB Gerät verbunden.\n\nUm fortzufahren, bitte Verbindung herstellen.\n\nWenn eins bereits IS verbunden ist, versuchen Sie es zu trennen und wieder anzuschließen. Das Kabel kann lose sein.
Toolbox.ArcticFoxConfiguration.Message.UnableToReadData=Die Geräteeinstellungen können nicht heruntergeladen werden. Schließen Sie das Gerät erneut an.

Toolbox.ArcticFoxConfiguration.Status.Device=Gerät ist
Toolbox.ArcticFoxConfiguration.Status.Connected=verbunden
Toolbox.ArcticFoxConfiguration.Status.Disconnected=nicht verbunden
Toolbox.ArcticFoxConfiguration.Status.Ready=Fertig
Toolbox.ArcticFoxConfiguration.Status.OperationCompleted=Vorgang abgeschlossen

Toolbox.ArcticFoxConfiguration.ActiveProfileLabel=Aktiv:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Akku 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Akku 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Akku 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Akku 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Bearbeiten
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Akku Modell:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=Prüfe TCR:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.LSLLabel=Leichter Schlafmodus:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Puff Cut-Off:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=RCOBC:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Shunt Korrektur:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=USB Laden:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=USB nicht schlafen:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.X32Label=X32:
Toolbox.ArcticFoxConfiguration.AdvancedTab=Fortgeschritten
Toolbox.ArcticFoxConfiguration.BVOTab=BVO
Toolbox.ArcticFoxConfiguration.CircleScreenTab=Wiederkehrender Bildschirm
Toolbox.ArcticFoxConfiguration.ClassicScreenTab=Klassischer Bildschirm
Toolbox.ArcticFoxConfiguration.ConfigurationButton=Konfiguration▾
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.New=Neu
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.Open=Öffnen
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.SaveAs=Speichern unter
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=Verbindung herstellen und herunterladen
Toolbox.ArcticFoxConfiguration.ControlsTab=Bedienelemente
Toolbox.ArcticFoxConfiguration.DownloadButton=Herunterladen
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=Aktiviert
Toolbox.ArcticFoxConfiguration.General.Controls.2ClicksLabel=2 Klicks:
Toolbox.ArcticFoxConfiguration.General.Controls.3ClicksLabel=3 Klicks:
Toolbox.ArcticFoxConfiguration.General.Controls.4ClicksLabel=4 Klicks:
Toolbox.ArcticFoxConfiguration.General.Controls.Power1WStepLabel=Leistung 1W Schritt:
Toolbox.ArcticFoxConfiguration.General.Controls.SharedLabel=Geteilt:
Toolbox.ArcticFoxConfiguration.General.Controls.TCModeClicksLabel=TC Modus Feuertaste Multi-Klicks:
[... 216 lines cut ...]
Toolbox.FirmwareUpdater.HardwareVersionLabel=Hardware version:
Toolbox.FirmwareUpdater.BootModeLabel=Boot mode:
Toolbox.FirmwareUpdater.ProgressGroupLabel=Progress:
Toolbox.FirmwareUpdater.OperationsGroupLabel=Operations:
Toolbox.FirmwareUpdater.CommonTab=Common
Toolbox.FirmwareUpdater.UpdateButton=Update from file
Toolbox.FirmwareUpdater.DatalfashTab=Dataflash
Toolbox.FirmwareUpdater.ReadDataflashButton=Read dataflash
Toolbox.FirmwareUpdater.WriteDataflashButton=Write dataflash
Toolbox.FirmwareUpdater.ResetDataFlashButton=Reset dataflash
Toolbox.FirmwareUpdater.AdvancedTab=Advanced
Toolbox.FirmwareUpdater.ChangeHWVerButton=Change HW Version
Toolbox.FirmwareUpdater.SwitchBootModeButton=Switch boot mode
Toolbox.FirmwareUpdater.CancelButton=Cancel

Toolbox.FirmwareUpdater.Messages.DeviceIsReady=Device is ready.
Toolbox.FirmwareUpdater.Messages.WaitingForDevice=Waiting for device...
Toolbox.FirmwareUpdater.Messages.WaitingForDeviceAfterReset=Waiting for device after reset...
[... cut ...] dataflash...
Toolbox.FirmwareUpdater.Messages.WritingDataflash=Writing dataflash...
Toolbox.FirmwareUpdater.Messages.RestartingDevice=Restarting device...
Toolbox.FirmwareUpdater.Messages.UploadingFirmware=Uploading firmware...
Toolbox.FirmwareUpdater.Messages.DataflashReadAndSave=Dataflash was successfully read and saved to the file.
Toolbox.FirmwareUpdater.Messages.DataflashWritten=Dataflash was successfully written.
Toolbox.FirmwareUpdater.Messages.DataflashReseted=Dataflash has been reseted.

TBXin, 2017-01-22:
Did you test your translation?
Some labels are too large.

[image]

BigBadBlackWolf, 2017-01-22:
Yepp, I saw. I have testet it not at all. I corrected it nightly. Sorry!

TBXin, 2017-01-22:
No problem, will wait for an update.

ArionWT, 2017-01-24:
Remains to cut down words' length there:
[image]
[image]
[image]

smok-e, 2017-01-24:
I've seen these. I will try to shorten the words.
But it is not easy to find words that are right and express what is needed there and are short enough too. I will do my best ;-) I will upload a new version tonight.

smok-e, 2017-01-24:
I tried to erase all mistakes I made and fit it to the Toolbox boundaries now.
I hope it is ok now.
[DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/727512/DE.lpack.txt)

TBXin, 2017-01-24:
@smok-e Thanks for your work, but i found some unfixed issues

[image]

[image]

[image]

[image]

ArionWT, 2017-01-24:
@TBXin on my way all looks ok 😐

smok-e, 2017-01-24:
??? I don't understand this, because it din't look like this here on my PC.
I will have a look into it right now.
I hope I can fix it asap.

TBXin, 2017-01-24:
@smok-e my bad, sorry.

smok-e, 2017-01-24:
ok, then I think it is ok now :-)

TBXin, 2017-01-24:
Done!
https://github.com/TBXin/NFirmwareEditor/commit/367f4c9c03ba2f550be035639e59eb4c157816fb

smok-e, 2017-01-24:
:-))

ArionWT, 2017-01-24:
Last one:
[image]

smok-e, 2017-01-24:
I found some more mistakes ;-)
here is the updated file:
[DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/727713/DE.lpack.txt)

I hope this is the last revision ;-)

smok-e, 2017-01-24:
I updated the file again, and edited my last post with the most recent file, so please download it again,
if you have downloaded it already

smok-e, 2017-01-24:
I've done e few more edits to the language pack, now everything should be really ok ;-)

[DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/727996/DE.lpack.txt)

smok-e, 2017-01-25:
I just found some more mistakes, and updated the language pack again.

[DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/730975/DE.lpack.txt)

XXMiToXX, 2017-01-26:
Edit German LNG file in different lines due to errors or obscurity.
[DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/732120/DE.lpack.txt)

smok-e, 2017-01-26:
I updated the german language pack now with some edits from the last update and some minor enhancements. Should be fine now. ;-)

[DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/732921/DE.lpack.txt)

XXMiToXX, 2017-01-27:
[image]

smok-e, 2017-01-27:
"_**Lade von Box” is not the best wording. It should be “Setup laden”
So, NOW it is perfect... :) (incl. NEW Cancel Button in Screenshooter**)_"

I don't agree, because I think "Setup laden" is not clear enough,
it cold be the setup from the device or from somewhere else (maybe a saved setup), 
that's why I used "Lade von Box"
even better would be "Setup vom Gerät laden" but the string isn't long enough.

So I would use either "Lade von Box" or "Lade vom Gerät".

SEEKNDESTROY, 2017-01-27:
"Lade vom Gerät" makes more sense to me.

smok-e, 2017-01-27:
So I did an update with "Setup von Gerät laden" I think this is the best wording,
but I don't mind if someone will use another wording. ;-)

[27-01-17--nightly--DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/736293/27-01-17--nightly--DE.lpack.txt)

BigBadBlackWolf, 2017-01-28:
Found some mistakes (wrong titles => new titles)

// Shared
Toolbox.Shared.SecondString=Sekunde
Toolbox.Shared.SecondsString=Sekunden

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox konfigurieren
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic konfigurieren

Toolbox.MainWindow.DeviceMonitorButton=Box Monitor
(If in all other textposition device named "Gerät" then we must do it for
all. One German titles for one and the same name. Box Monitor => Geräte
Monitor)

Toolbox.MainWindow.ScreenshooterButton=Bildschirmfotos
Toolbox.MainWindow.FirmwareUpdaterButton=Firmware Updater
Toolbox.MainWindow.AboutLink=über && Links

Toolbox.MainWindow.Tray.ShowToolbox=Zeige NFE Toolbox Fenster
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Öffne
ArcticFox Konfiguration sobald Gerät angeschlossen wird.
Toolbox.MainWindow.Tray.TimeSync=Syncronisiere Uhrzeit wenn Gerät verbunden
ist
Toolbox.MainWindow.Tray.StartWithWindows=Starte NFE Toolbox wenn Windows
startet
Toolbox.MainWindow.Tray.Exit=Beenden

// ArcticFox Configuration Window
Toolbox.ArcticFoxConfiguration.Message.ConnectDevice=Verbinde Gerät:
\n\nerwarte ArcticFox Firmware\nV [{0}]\n\noder neuere Version.
Toolbox.ArcticFoxConfiguration.Message.DownloadingSettings=Lade von Gerät...
Toolbox.ArcticFoxConfiguration.Message.OutdatedToolbox=NFE Toolbox ist
veraltet.\n\nUm fortzufahren laden Sie bitte\n\ndie neueste Version.

Toolbox.ArcticFoxConfiguration.Message.NoCompatibleUSBDevice=Kein passendes
Gerät verbunden.\n\nZum fortfahen, verbinden Sie bitte ein Gerät.\n\nWenn
Sie bereits ein Gerät angeschlossen haben, trennen Sie es und verbinden es
wieder. Sollte das Gerät dann noch nicht erkannt werden, prüfen Sie das
USB-Kabel.
(Zum Fortfahren => Zum Fortfahren)

Toolbox.ArcticFoxConfiguration.Message.UnableToReadData=Kann Einstellungen
nicht laden. Verbinden Sie das Gerät erneut.

Toolbox.ArcticFoxConfiguration.Status.Device=Gerät ist
Toolbox.ArcticFoxConfiguration.Status.Connected=verbunden
Toolbox.ArcticFoxConfiguration.Status.Disconnected=nicht verbunden
Toolbox.ArcticFoxConfiguration.Status.Ready=Fertig
Toolbox.ArcticFoxConfiguration.Status.OperationCompleted=Vorgang beendet

Toolbox.ArcticFoxConfiguration.ActiveProfileLabel=Aktiv:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Batterie 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Batterie 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Batterie 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Batterie 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Einstellen

Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Batterie
Model:
(Batterie Model: => Batterie Modell:)

Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=Prüfe TCR:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.LSLLabel=LSL Modus:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Zug
Cut-Off:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=RCOBC:

Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Shunt
Korrektur:
[... 316 lines cut ...]
Toolbox.FirmwareUpdater.UpdateButton=Update aus Datei
Toolbox.FirmwareUpdater.DatalfashTab=Dataflash
Toolbox.FirmwareUpdater.ReadDataflashButton=Lese dataflash
Toolbox.FirmwareUpdater.WriteDataflashButton=Schreibe dataflash
Toolbox.FirmwareUpdater.ResetDataFlashButton=Setze dataflash zurück
Toolbox.FirmwareUpdater.AdvancedTab=Experte
Toolbox.FirmwareUpdater.ChangeHWVerButton=Ändere HW Version
Toolbox.FirmwareUpdater.SwitchBootModeButton=Wechsel Boot Art
Toolbox.FirmwareUpdater.CancelButton=Zurück

Toolbox.FirmwareUpdater.Messages.DeviceIsReady=Gerät ist fertig.
Toolbox.FirmwareUpdater.Messages.WaitingForDevice=Warte auf Gerät...
Toolbox.FirmwareUpdater.Messages.WaitingForDeviceAfterReset=Warte auf Gerät
nach zurücksetzen...
Toolbox.FirmwareUpdater.Messages.ReadingDataflash=Lese dataflash...
Toolbox.FirmwareUpdater.Messages.WritingDataflash=Schreibe dataflash...
Toolbox.FirmwareUpdater.Messages.RestartingDevice=Starte Gerät neu...
Toolbox.FirmwareUpdater.Messages.UploadingFirmware=Lade Firmware auf
Gerät...
Toolbox.FirmwareUpdater.Messages.DataflashReadAndSave=Dataflash wurde
erfolgreich gelesen und in Datei gespeichert.
Toolbox.FirmwareUpdater.Messages.DataflashWritten=Dataflash wurde
erfolgreich geschrieben.
Toolbox.FirmwareUpdater.Messages.DataflashReseted=Dataflash wurde
zurückgesetzt.

smok-e, 2017-01-28:
I agree with allmost all the enhancements you did,
the only thing I'm not shure  is the dominant = Vorgabe,
I don't think it has really the same meaning.
So I kept dominant. Maybe it could be something like 
"Temperatur bestimmt" or "Temperatur beherrscht" ?

I really don't know what the meaning of this is:

Toolbox.ArcticFoxConfiguration.InfoLines.NonDominant=Nicht dominant L/T
(nicht dominant L/T
What is mean with this term?
=> Keine Vorgabe L/T)

so I just tried to translate it.
Maybe "Keine Vorgabe..." can be correct,
or maybe " nicht L/T bestimmt"?

here is the updated language pack:

[28-01-17--DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/736906/28-01-17--DE.lpack.txt)

smok-e, 2017-01-28:
soemthing seems to be wrong with the latest uopload I did,
it didn't have the changes included,
so I upload it again now with the changes 
[28-01-17-2--DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737194/28-01-17-2--DE.lpack.txt)

smok-e, 2017-01-31:
[image]

I just found another issue.
I changed the translation to fit,
but I'm not sure that this is really correct translated.
I Would like to ask other german users to help here.

[31-01-2017--DE.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/743251/31-01-2017--DE.lpack.txt)

I also would like to translate 
"Preheat..." to "Anheiz... "
like:
 "Preheat curve" to "Anheizkurve",
"Preheat Delay" to "Anheiz Pause" ("Anheizpause"),
"Preheat Power" to "Anheiz Leistung"("Anheizleistung"),
"Preheat Type" to "Anheiz Art",
"Preheat Time" to "Anheiz Zeit"

what do other german users think?

BigBadBlackWolf, 2017-02-01:
@Stephan
The translation for "preheat" correctly is "Vorheizen", but I think
"Anheizen" is the better word.

"Preheat curve" to "Anheizkurve",
"Preheat Delay" to "Anheizpause",
"Preheat Power" to "Anheizleistung",
"Preheat Type" to "Anheizart",
"Preheat Time" to "Anheizzeit"

XXMiToXX, 2017-02-01:
> @Stephan The translation for "preheat" correctly is "Vorheizen", but I think "Anheizen" is the better word. "Preheat curve" to "Anheizkurve", "Preheat Delay" to "Anheizpause", "Preheat Power" to "Anheizleistung", "Preheat Type" to "Anheizart", "Preheat Time" to "Anheizzeit" 

"Vorheizen" is the best wording to explain "Preheat"

Best regards

TBXin, 2017-02-01:
@XXMiToXX 
1. do not spam in the any related issue your thoughts;
2. i will not redesign UI in the near future;

XXMiToXX, 2017-02-01:
> @XXMiToXX

   1. do not spam in the any related issue your thoughts;

TBXin, 2017-02-01:
> And WOW, the UI looks good for large words or sentence.

As usual. You do not see the whole situation, so you all seem simple and obvious.

> The are some points that have to edit also, but it is not really difficult.

I said that I would not change the layout.

XXMiToXX, 2017-02-01:
> As usual. You do not see the whole situation, so you all seem simple and obvious.

I know, as usual.
Thanks for your support. Think about it.....

smok-e, 2017-02-01:
@BigBadBlackWolf : thank you for the support :-)

@ Mirko (XXMiToXX): I know that "Vorheizen" is the best translation for Preheat,
but I think it is not really preheating = Vorheizen, as it is a change in the heating curve, and not happening before the heating,
so I think "Anheizen" is the better german word for this, as in my understanding it means:
the beginning of the heating.

and your shunt correction translation = "Ohm/Widerstand Korrektur" is correct, but too long for the string, as you see in the picture,
so I think it is better to keep just "Ohm Korrektur"
[image]

if you agree, I will upload a new language pack version tonight

smok-e, 2017-02-01:
ps.: I also would like some GUI changes,
but it is totally up to TBXin, and if he doesn't want it,
that's fine by me, and I will edit the translation to fit the GUI as it is.

I really apprecciate all the work that he does to make the firmeware working,
and everything for free, that is more than I could ever demand.
So I'm really happy, when I can contribute just a little bit, 
to express my thankfulness. jm2c

XXMiToXX, 2017-02-02:
> and your shunt correction translation = "Ohm/Widerstand Korrektur" is correct, but too long for the string...
Your' right. Agree.
But not to "Anheizen" :) it is like: "light the fire in the fireplace" ;)

> but it is totally up to TBXin, and if he doesn't want it...
Yes it is!
But it is no problem to edit it. It's only a UI. No changes to the firmware. 
I already download source and edit it. It works for my test devices (Wismec and Eleaf) 

> I really apprecciate all the work that he does to make the firmeware working, and everything for free, that is more than I could ever demand. So I'm really happy, when I can contribute just a little bit...
Me to, great WORK!!
So i don't talk about the UI ;)

Let us find the best translation...

smok-e, 2017-02-02:
I still think "Anheizen" is the best working translation
for the GUI

here is the meaning from the Duden:
BEDEUTUNGSÜBERSICHT	ℹ
(in einem Ofen o. Ä.) ein Feuer entfachen, zu heizen beginnen
(umgangssprachlich) steigern, größer, heftiger, stärker werden lassen; schüren

and I think this is exactly what preheat means

So I would use the wording as I suggested,
and we don't need to edit the GUI ;-)

TBXin, 2017-02-02:
Guys, i appreciate the work you are doing for the localization of the NFE.
I would like to ask you to make changes in the localization file through pull requests.
I can merge pull requests with one click, so that will be much faster than at this point ^^

smok-e, 2017-02-02:
ok, I hope I did in the right way ;-)
I'm a newbie to github

TBXin, 2017-02-02:
@smok-e oh, almost :) First of all you should take latest repository and only than moidify it and create changes. Your pull request contains 40 file changes. I can't apply it now, so i will discard it and will wait for pull request only with single file change.

smok-e, 2017-02-04:
??? how and where do I find the last repository? and how do I modify it?
Sorry, but I've never done something like this.

BigBadBlackWolf, 2017-02-14:
Since several questions were asked, why the myEvic button doesn't work in NFE Toolbox, the language packages should be adapted by a small change.
Here is my suggestion for the German language pack:
// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox konfigurieren
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic konfigurieren\n(derzeit inaktiv)

and for the English language pack:
// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox Configuration
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic Configuration\n(not supported yet)

All other language packages please also adjust.

Or we leave this button completely away, as long as myevic is not supported. If this is the case, it can be inserted again. Thus, no requests are made by users why this button does not work.