TBXin/NFirmwareEditor #144: Polish Language Pack for NToolbox

bobsilesia, 2017-01-25:
Hi,
Here is my Polish Translation for NToolbox software.
Thanks a lot for your software. :)

Best Regards
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/728362/PL.lpack.txt)

bobsilesia, 2017-01-26:
Hello,
I posted some corrections in translation.
Thank you again, great software. 

Best regards,
bobsilesia
[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/733238/PL.lpack.txt)

bobsilesia, 2017-01-26:
Hi,
and one small patch

Best regards,
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/733344/PL.lpack.txt)

bobsilesia, 2017-01-27:
Hello,
last version of LP_pl

Best regards,
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/736036/PL.lpack.txt)

bobsilesia, 2017-01-28:
Hi,
a few changes...

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737011/PL.lpack.txt)

Best regards,
bobsilesia

bobsilesia, 2017-01-28:
Hi,
I think that this is the final version LP_pl. :)
Thank you very much for this unique software.

Best regards,
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737089/PL.lpack.txt)

bobsilesia, 2017-01-29:
Hi,
It's probably all.

Best regards,
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737384/PL.lpack.txt)

bobsilesia, 2017-01-29:
Hi,
sorry... small patch.

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737401/PL.lpack.txt)

TBXin, 2017-01-29:
Leave it open, for further improvements.

bobsilesia, 2017-01-29:
Hi,
you're right :)

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737561/PL.lpack.txt)

bobsilesia, 2017-01-29:
Hi,
some corrections.

All the best,
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737897/PL.lpack.txt)

bobsilesia, 2017-01-31:
@TBXin 
Hi,
some corrections.

All the best,
bobsilesia

[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/741313/PL.lpack.txt)

barsuc, 2017-01-31:
Poprawiłem słowo "Ściągnij" na "Pobierz", lepiej brzmi.
I corrected word "Ściągnij", now is "Pobierz". I think it's better. ;)

For the Greater Good. 
barsuc
[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/741957/PL.lpack.txt)

bobsilesia, 2017-01-31:
ok,
let it be "Pobierz".

Best regards,
bobsilesia

bobsilesia, 2017-01-31:
Hi,
On this occasion. A small patch ...
[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/742511/PL.lpack.txt)

@barsuc 
Thanks :)

Regards,
bobsilesia

barsuc, 2017-01-31:
My pleasure ;)
For the Greater Good.
barsuc

Rozpoznany, 2017-05-09:
Hi,
last version of LP_pl for new NF 2.0 
[PL.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/985163/PL.lpack.txt)

Best regards, rozpozany

TBXin, 2017-05-09:
@Rozpoznany good job 👍