TBXin/NFirmwareEditor #148: Missing button in "Screenshooter"

XXMiToXX, 2017-01-27:
Hi,

is not an issue, it is only a "nice to have" option....

In the "Firmware updater" is a Cancel button, if you click this button it goes back to Main GUI.
In the "Screenshooter" there is NO button like is. You have to click the "X". ;)

Best regards
Mirko

TBXin, 2017-01-27:
> In the "Screenshooter" there is NO button like is. You have to click the "X". ;)

Or you can just press Esc button ^^

XXMiToXX, 2017-01-27:
> Or you can just press Esc button ^^

Yes, if you knows that it works. But the button is intuitive and user friendly... ;)

TBXin, 2017-01-27:
Done in the nightly builds.

[image]

Daveee10, 2017-01-29:
Just a small thing, but IMO the Close button instead of Cancel would better.

closed 2017-02-03 by TBXin