TBXin/NFirmwareEditor #149: Box Monitor in 6.4.62 is weird with my iStickTC100W

smok-e, 2017-01-27:
I 
The Box Monitor seems a bit confused.
It shows a Temperature of 36°C - 40°C (red line) when I take a puff, and the TC is set to 240°C (brown line)

and when I rescale the GUI of the box monitor (make it bigger ) the coloured lines go over the boxes.

maybe it is only my iStick and PC-usb connection, that is working weird.
But everything else is working fine.

TBXin, 2017-01-27:
Can you attach a screenshot of this behavior?

smok-e, 2017-01-27:
[image]
this is the screen with the blurred lines

and now the qweird temperature:
[image]

I hope this explains what I meant

TBXin, 2017-01-28:
@smok-e test, please, [this build](https://ci.appveyor.com/project/TBXin/nfirmwareeditor/build/6.4.64/artifacts).

BigBadBlackWolf, 2017-01-28:
@ReikoKitsune look last pic "NFE Toolbox - Device Monitor" at this thread
from Stephan there are wrong words in Germany Translation from DE.Ipack.txt.

// Device Monitor Window
...
Toolbox.DeviceMonitor.OutputCurrentLabel=Ausgangsspannung:
Toolbox.DeviceMonitor.OutputVoltageLabel=Ausgansvolt:

OutputCurrent is Current not Voltage, also the correct word in German is
"Ausgangsstrom:"
OutputVoltage is Voltage, also the correct word in German is
"Ausgangsspannung:"

Toolbox.DeviceMonitor.RecordButton=Aufnehmen...
Toolbox.DeviceMonitor.RecordStopButton=Stop Aufnahme

Grammatic Correction
RecordButton=Start Aufnahme...
RecordStopButton=Stopp Aufnahme

smok-e, 2017-01-28:
I tested with the 6.4.65

the blurred lines are gone, but the recording is still weird:

[image]

TBXin, 2017-01-28:
@smok-e your coil material?

smok-e, 2017-01-28:
Sorry, I think it was my mistake.
I think you're right, I don't remember the material I put into the atomizer,
but I changed the atomizer, and now the recording is working ok with SS304.

So maybe it would be helpfull, if there could be an error message when someone does 
soemthing like I tried. 
Maybe like: "Which material is the coil? Kanthal shows wrong values in TCR mode" or something like this. But maybe it is too complicated to programm it into the software?

TBXin, 2017-01-28:
Hm, you yourself disabled this kind of errors. Menu -> Settings -> Expert -> CheckTCR.
By default mod will stop fire Kanthal in TC mode and shows you "TCR Error" message.

smok-e, 2017-01-28:
yes, on the box it is working this way, but I meant the "box monitor",
can it show the same message? I think it would be  helpfull.

closed 2017-02-01 by TBXin