TBXin/NFirmwareEditor #151: [Language Pack] NToolbox - Serbian

ivanneko, 2017-01-28:
Hi,
 Here is my Serbian Translation for NToolbox software.
 Thanks a lot for your software. :)

Best Regards
ivanneko
[SR.lpack.zip](https://github.com/TBXin/NFirmwareEditor/files/737165/SR.lpack.zip)

TBXin, 2017-01-28:
Thanks. I've added your localization. 
Please for the future - do not delete blank lines in the file.
I keep it for more comfortable navigation.

ivanneko, 2017-01-29:
I apologize for that will not happen again ..
Thanks

ivanneko, 2017-05-09:
Here is update for my Serbian Translation for NToolbox software.
Thanks
[SR.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/988192/SR.lpack.txt)

ivanneko, 2017-05-10:
Here is update for my Serbian Translation for NToolbox software.
Thanks
[SR.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/988386/SR.lpack.txt)