TBXin/NFirmwareEditor #158: [Language Pack] NToolbox - French

ZelTroN-2k3, 2017-01-29:
**Issue Descriptions:**
[Language Pack] NToolbox - French/Français

**Steps to Reproduce:**
[FR.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737606/FR.lpack.txt)

TBXin, 2017-01-29:
You should take existed black and modify it for your language.
Your file does not contains some localizable strings.

ZelTroN-2k3, 2017-01-29:
Hello, TBXin

This FR.lpack.txt file, and integrated into the scr / out / languages folder,
And works very well with me.

Your file does not contain some localizable strings ???
I do not understand you....

TBXin, 2017-01-29:
You file contains 281 lines, other files contains 306 lines.

[image]

ZelTroN-2k3, 2017-01-29:
Ok TBXin,
Sorry, I will do the correction of the missing chains.
thank you.

ZelTroN-2k3, 2017-01-29:
TBXin,
News, correction....

Issue Descriptions:
[Language Pack] NToolbox - French/Français

Steps to Reproduce:
News, correction Language Pack
[FR.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737635/FR.lpack.txt)

TBXin, 2017-01-29:
@ZelTroN-2k3 now you should test what you want :)

Some texts are too large to fit desired places.

[image]

[image]

ZelTroN-2k3, 2017-01-29:
TBXin,

I will reduce the field to the shortest, so that it is readable and understandable in French language.

TBXin, 2017-01-29:
@ZelTroN-2k3 do not rush and do everything accurately.

ZelTroN-2k3, 2017-01-29:
TBXin,
News, correction....
I will reduce the field to the shortest, so that it is readable and understandable in French language.

Issue Descriptions:
[Language Pack] NToolbox - French/Français

Steps to Reproduce:
News, correction Language Pack
[FR.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/737665/FR.lpack.txt)
[image]
[image]
[image]
[image]
[image]

TBXin, 2017-01-29:
Better, but...

[image]

[image]

[image]

ZelTroN-2k3, 2017-01-29:
TBXin,
News, correction....
**I will reduce the field to the shortest, so that it is readable and understandable in French language.**

Issue Descriptions:
[Language Pack] NToolbox - French/Français

Steps to Reproduce:
News, correction Language Pack

[**FR.lpack.txt**](https://github.com/TBXin/NFirmwareEditor/files/737690/FR.lpack.txt)

[image]
[image]

PS:
-> Activé  - EnabledCheckBox=Enabled -> **corrected**

TBXin, 2017-01-29:
@ZelTroN-2k3 you miss Pi-Regulator window.

Also preheat type power label starts with lower case.

[image]

ZelTroN-2k3, 2017-01-29:
TBXin,
News, correction....
**I will reduce the field to the shortest, so that it is readable and understandable in French language.**

Issue Descriptions:
[Language Pack] NToolbox - French/Français

Steps to Reproduce:
News, correction Language Pack

[**FR.lpack.txt**](https://github.com/TBXin/NFirmwareEditor/files/737705/FR.lpack.txt)

[image]

[image]

XXMiToXX, 2017-02-01:
@TBXin 
I think it is easier to redesign the GUI so that other languages have enough space... 
maybe 50px to the right... ;)

[image]

ZelTroN-2k3, 2017-02-01:
Hello TWin,
Yes that would be sensible, 50px right ok.
Its going to be simpler also for the other languages.

Cordially.
ZelTroN2k3

ZelTroN-2k3, 2017-02-01:
Hello XXMiToXX.

Good.
Cordially.
ZelTroN2k3