TBXin/NFirmwareEditor #163: [Language Pack] NToolbox - Dutch

Wietjuh, 2017-02-03:
Translation to Dutch:

//NL Translation by Wietjuh
//170123

// Shared
Toolbox.Shared.SecondString=seconde
Toolbox.Shared.SecondsString=seconden

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox Configuratie
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic Configuratie
Toolbox.MainWindow.DeviceMonitorButton=Apparaat Monitor
Toolbox.MainWindow.ScreenshooterButton=Schermopname
Toolbox.MainWindow.FirmwareUpdaterButton=Firmware Updater
Toolbox.MainWindow.AboutLink=Over && Links

Toolbox.MainWindow.Tray.ShowToolbox=Open NFE Toolbox venster
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Open ArcticFox Configuratie als het apparaat is verbonden
Toolbox.MainWindow.Tray.TimeSync=Synchronizeer de tijd als het Apparaat verbonden is
Toolbox.MainWindow.Tray.StartWithWindows=Open NFE Toolbox bij het opstarten van Windows
Toolbox.MainWindow.Tray.Exit=Sluiten

// ArcticFox Configuration Window
Toolbox.ArcticFoxConfiguration.Message.ConnectDevice=Verbind Apparaat met\n\nArcticFox\n[{0}]\n\nfirmware of nieuwer
Toolbox.ArcticFoxConfiguration.Message.DownloadingSettings=Instellingen downloaden...
Toolbox.ArcticFoxConfiguration.Message.OutdatedToolbox=NFE Toolbox is verouderd.\n\nOm verder te gaan downoad \n\nde meest recente versie.
Toolbox.ArcticFoxConfiguration.Message.NoCompatibleUSBDevice=Geen beschikbare USB Apparaten verbonden.\n\nOm verder tegaan, verbind een apparaat.\n\nAls deze al is verbonden, probeer dan opnieuw te verbinden.
Toolbox.ArcticFoxConfiguration.Message.UnableToReadData=Niet mogenlijk om apparaatinstellingen te downloaden, verbind uw apparaat opnieuw.

Toolbox.ArcticFoxConfiguration.Status.Device=Apparaat is
Toolbox.ArcticFoxConfiguration.Status.Connected=verbonden
Toolbox.ArcticFoxConfiguration.Status.Disconnected=niet verbonden
Toolbox.ArcticFoxConfiguration.Status.Ready=klaar
Toolbox.ArcticFoxConfiguration.Status.OperationCompleted=Opdracht voltooid

Toolbox.ArcticFoxConfiguration.ActiveProfileLabel=Actief:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Batterij 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Batterij 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Batterij 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Batterij 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Aanpassen
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Batterij model:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=Check TCR:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.LSLLabel=Lichte Slaapmodus:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Puff Cut-Off:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=RCOBC:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Shunt Correctie:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=USB Opladen:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=USB Niet Slapen:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.X32Label=X32:
Toolbox.ArcticFoxConfiguration.AdvancedTab=Geavanceerd
Toolbox.ArcticFoxConfiguration.BVOTab=BVO
Toolbox.ArcticFoxConfiguration.CircleScreenTab=Cirkel Thema
Toolbox.ArcticFoxConfiguration.ClassicScreenTab=Klassiek Thema
Toolbox.ArcticFoxConfiguration.ConfigurationButton=Configuratie▾
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.New=Nieuw
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.Open=Open
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.SaveAs=Opslaan als
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=Verbind en download configuratie
Toolbox.ArcticFoxConfiguration.ControlsTab=Bedienelementen
Toolbox.ArcticFoxConfiguration.DownloadButton=Download instellingen
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=Geactiveerd
Toolbox.ArcticFoxConfiguration.General.Controls.2ClicksLabel=2 Klikken:
Toolbox.ArcticFoxConfiguration.General.Controls.3ClicksLabel=3 Klikken:
Toolbox.ArcticFoxConfiguration.General.Controls.4ClicksLabel=4 Klikken:
Toolbox.ArcticFoxConfiguration.General.Controls.Power1WStepLabel=Power per 1W:
Toolbox.ArcticFoxConfiguration.General.Controls.SharedLabel=Gedeeld:
Toolbox.ArcticFoxConfiguration.General.Controls.TCModeClicksLabel=TC Modus Fire Button Multi Kliks:
Toolbox.ArcticFoxConfiguration.General.Controls.UpDownButtonsLabel=Op/Neer knoppen:
[... 214 lines cut ...]
Toolbox.FirmwareUpdater.HardwareVersionLabel=Hardware versie:
Toolbox.FirmwareUpdater.BootModeLabel=Opstartmodus:
Toolbox.FirmwareUpdater.ProgressGroupLabel=Voortgang:
Toolbox.FirmwareUpdater.OperationsGroupLabel=Taken:
Toolbox.FirmwareUpdater.CommonTab=Algemeen
Toolbox.FirmwareUpdater.UpdateButton=Update met een bestand
Toolbox.FirmwareUpdater.DatalfashTab=Dataflash
Toolbox.FirmwareUpdater.ReadDataflashButton=Lees dataflash
Toolbox.FirmwareUpdater.WriteDataflashButton=Schrijf dataflash
Toolbox.FirmwareUpdater.ResetDataFlashButton=Reset dataflash
Toolbox.FirmwareUpdater.AdvancedTab=Geavanceerd
Toolbox.FirmwareUpdater.ChangeHWVerButton=Verander HW Versie
Toolbox.FirmwareUpdater.SwitchBootModeButton=Wissel opstartmodus
Toolbox.FirmwareUpdater.CancelButton=Annuleren

Toolbox.FirmwareUpdater.Messages.DeviceIsReady=Apparaat is klaar voor gebruik...
Toolbox.FirmwareUpdater.Messages.WaitingForDevice=Wachten op het apparaat...
Toolbox.FirmwareUpdater.Messages.WaitingForDeviceAfterReset=Wachten op apparaat na een reset...
Toolbox.FirmwareUpdater.Messages.ReadingDataflash=Leest dataflash...
Toolbox.FirmwareUpdater.Messages.WritingDataflash=Schrijft dataflash...
Toolbox.FirmwareUpdater.Messages.RestartingDevice=Opnieuw opstarten van het apparaat...
Toolbox.FirmwareUpdater.Messages.UploadingFirmware=Uploaden van firmware...
Toolbox.FirmwareUpdater.Messages.DataflashReadAndSave=Dataflash is succesvol uitgelezen en opgeslagen naar een bestand.
Toolbox.FirmwareUpdater.Messages.DataflashWritten=Dataflash is succesvol weggeschreven.
Toolbox.FirmwareUpdater.Messages.DataflashReseted=Dataflash is gereset.

Wietjuh, 2017-02-03:
Tested and functional!

TBXin, 2017-02-03:
> Tested

Don't think so.

[image]

[image]

And many more places where labels are too large.

Wietjuh, 2017-02-03:
That i havent done yet, working on it

Wietjuh, 2017-02-04:
Woops... I see i've uploaded the incorrect text. Updated and doublechecked

TBXin, 2017-02-09:
Required a few more fixes.

[image]

[image]