TBXin/NFirmwareEditor #164: Popup dialog focus in fw updater

joeblowma, 2017-02-04:
When using firmware updater, the "Firmware successfully updated." popup sometimes shows up behind the updater window. I was force closing via task manager until I realized this was the case ^^.

This might only be an issue on windows 10, where they seem to have thrown out the rules for rendering new dialogs above whatever was in focus and created it.

Reproduce:
Flash a firmware with the updater, even if the dialog does initially appear above you can push it behind with the taskbar by selecting the program instead of the dialog. Windows seems to be too inconsistent and (seemingly) randomly pushes the dialog behind on it's own. Google shows this to be an issue in general with C# where dialogs outside the form get created.

Possible solution (probably not the best one, c# isn't really my thing), in InfoBox.cs insert
`new Form { TopMost = true },`
into the two MessageBox.Show commands - this makes the popup a "part" of the main window via an invisible form and does not create a separate taskbar item for the popup. Not sure if this would work well for other messages though.
ie:
`MessageBox.Show(new Form { TopMost = true }, text, @"Information", MessageBoxButtons.OK, MessageBoxIcon.Information);`

TBXin, 2017-02-04:
Hi,

I know about this issue. I will do my best^^ But i don't like approach you are talking about, namely, I don't like that each message box will produce additional form, even if they are hidden.

joeblowma, 2017-02-04:
I was looking for some kind of BringToFront() but then I remembered... it's C#, it can't be that easy xD

Does the hidden form give a penalty? Memory leakage? Invisible boundaries you can't render outside of?

TBXin, 2017-02-05:
Every form is a windows resources, unmanaged resources, like handles and many other things.
Creating dummy objects is not a good approach to solve problems ^^

I can try to solve this problem another way, but it difficult to test, because i can't tell what i should do to reproduce this windows behavior.

joeblowma, 2017-02-06:
I'd think if you can solve the problem of the blocking dialog (blocking as in you can't close the flasher or hit cancel until the dialog is dismissed) being able to go behind the flashing window, perhaps by eliminating the extra task bar item by tieing the dialog to that window's form or instance then the randomness of it appearing behind would go away.

I haven't been able to make it happen on demand as of yet, I've flashed a bunch of times tinkering with a font and it always catches me by surprise when it happens, even knowing now that it does happen ^^

example (again, c# isn't my thing so perhaps there are downsides to this that I can't see)
in infobox.cs I added another method:
`public static void Show(IWin32Window owner, string text)
 {
MessageBox.Show(owner, text, @"Information", MessageBoxButtons.OK, MessageBoxIcon.Information);
}`
in FirmwareUpdaterWindow.cs I changed the call to:
`if (isSuccess)
{
InfoBox.Show(this.FindForm(), "Firmware successfully updated.");
}`
And that effectively attaches the dialog to the window without creating a dummy form or a taskbar item, and has the benefit of not needing to be applied universally to everything that uses show()

TBXin, 2017-02-06:
Some little changes to the code ^^ Now you able to test it :P

joeblowma, 2017-02-07:
"Little" ^^ Build 6.5.16 from AppVeyor no longer has the issue. Thanks once again TBXin :)

closed 2017-02-07 by joeblowma