TBXin/NFirmwareEditor #171: [Language Pack] NToolbox - Italian

mirko93s, 2017-02-08:
// Shared
Toolbox.Shared.SecondString=secondo
Toolbox.Shared.SecondsString=secondi

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=Configurazione ArcticFox
Toolbox.MainWindow.MyEvicConfigurationButton=Configurazione myEvic
Toolbox.MainWindow.DeviceMonitorButton=Monitoraggio dispositivo
Toolbox.MainWindow.ScreenshooterButton=Istantanea Schermo
Toolbox.MainWindow.FirmwareUpdaterButton=Aggiornamento Firmware
Toolbox.MainWindow.AboutLink=Informazioni && Link

Toolbox.MainWindow.Tray.ShowToolbox=Mostra finestra NFE Toolbox
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Apri Configuratore ArcticFox con il dispositivo connesso.
Toolbox.MainWindow.Tray.TimeSync=Sincronizza orologio quando il dispositivo è collegato
Toolbox.MainWindow.Tray.StartWithWindows=Apri NFE Toolbox all'avvio di Windows
Toolbox.MainWindow.Tray.Exit=Esci

// ArcticFox Configuration Window
Toolbox.ArcticFoxConfiguration.Message.ConnectDevice=Connetti dispositivo con\n\nArcticFox\n[{0}]\n\nfirmware o più recente
Toolbox.ArcticFoxConfiguration.Message.DownloadingSettings=Download impostazioni...
Toolbox.ArcticFoxConfiguration.Message.OutdatedToolbox=NFE Toolbox non è aggiornato.\n\nPer proseguire, per favore scarica\n\nl'ultima versione disponibile.
Toolbox.ArcticFoxConfiguration.Message.NoCompatibleUSBDevice=Non è connessuno alcun dispositivo USB compatibile.\n\nPer proseguire, connettine uno.\n\nSe è già connesso, prova a scllegarlo e collegarlo di nuovo. Il cavo potrebbe essere lento.
Toolbox.ArcticFoxConfiguration.Message.UnableToReadData=Impossibile scaricare le impostazioni del dispositivo. Riconnetti il tuo dispositivo.

Toolbox.ArcticFoxConfiguration.Status.Device=Il dispositivo è
Toolbox.ArcticFoxConfiguration.Status.Connected=connesso
Toolbox.ArcticFoxConfiguration.Status.Disconnected=disconnesso
Toolbox.ArcticFoxConfiguration.Status.Ready=Pronto
Toolbox.ArcticFoxConfiguration.Status.OperationCompleted=Operazione completata

Toolbox.ArcticFoxConfiguration.ActiveProfileLabel=Attivo:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Batteria 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Batteria 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Batteria 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Batteria 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Modifica
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Modello Batteria:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=Controlla TCR:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.LSLLabel=Light Sleep Mode:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Limitatore Tiro:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=RCOBC:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Correzione Shunt:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=Ricarica USB:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=Nessun Risparmio:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.X32Label=X32:
Toolbox.ArcticFoxConfiguration.AdvancedTab=Avanzato
Toolbox.ArcticFoxConfiguration.BVOTab=BVO
Toolbox.ArcticFoxConfiguration.CircleScreenTab=Tema Cerchio
Toolbox.ArcticFoxConfiguration.ClassicScreenTab=Tema Classico
Toolbox.ArcticFoxConfiguration.ConfigurationButton=Configurazione▾
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.New=Nuovo
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.Open=Apri
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.SaveAs=Salva con nome
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=Connetti e Scarica configurazione
Toolbox.ArcticFoxConfiguration.ControlsTab=Controlli
Toolbox.ArcticFoxConfiguration.DownloadButton=Scarica impostazioni
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=Abilitato
Toolbox.ArcticFoxConfiguration.General.Controls.2ClicksLabel=2 Click:
Toolbox.ArcticFoxConfiguration.General.Controls.3ClicksLabel=3 Click:
Toolbox.ArcticFoxConfiguration.General.Controls.4ClicksLabel=4 Click:
Toolbox.ArcticFoxConfiguration.General.Controls.Power1WStepLabel=Potenza step 1W:
Toolbox.ArcticFoxConfiguration.General.Controls.SharedLabel=Condiviso:
Toolbox.ArcticFoxConfiguration.General.Controls.TCModeClicksLabel=Modalità TC Multi Click Pulsante Fire:
Toolbox.ArcticFoxConfiguration.General.Controls.UpDownButtonsLabel=Pulsanti Su/Giù:
Toolbox.ArcticFoxConfiguration.General.Controls.VWModeClicksLabel=Modalità VW Multi Click Pulsante Fire:
Toolbox.ArcticFoxConfiguration.General.Controls.WakeUpPlusMinusLabel=Risveglia con + / - :
Toolbox.ArcticFoxConfiguration.General.Layout.Line1Label=Linea 1:
Toolbox.ArcticFoxConfiguration.General.Layout.Line2Label=Linea 2:
Toolbox.ArcticFoxConfiguration.General.Layout.Line3Label=Linea 3:
[... 211 lines cut ...]
Toolbox.FirmwareUpdater.HardwareVersionLabel=Versione hardware:
Toolbox.FirmwareUpdater.BootModeLabel=Modalità di boot:
Toolbox.FirmwareUpdater.ProgressGroupLabel=Progresso:
Toolbox.FirmwareUpdater.OperationsGroupLabel=Operazioni:
Toolbox.FirmwareUpdater.CommonTab=Comune
Toolbox.FirmwareUpdater.UpdateButton=Aggiorna da file
Toolbox.FirmwareUpdater.DatalfashTab=Dataflash
Toolbox.FirmwareUpdater.ReadDataflashButton=Leggi dataflash
Toolbox.FirmwareUpdater.WriteDataflashButton=Scrivi dataflash
Toolbox.FirmwareUpdater.ResetDataFlashButton=Resetta dataflash
Toolbox.FirmwareUpdater.AdvancedTab=Avanzato
Toolbox.FirmwareUpdater.ChangeHWVerButton=Cambia versione HW
Toolbox.FirmwareUpdater.SwitchBootModeButton=Cambia modalità boot
Toolbox.FirmwareUpdater.CancelButton=Annulla

Toolbox.FirmwareUpdater.Messages.DeviceIsReady=Dispositivo pronto.
Toolbox.FirmwareUpdater.Messages.WaitingForDevice=In attesa del dispositivo...
Toolbox.FirmwareUpdater.Messages.WaitingForDeviceAfterReset=Attesa del dispositivo dopo il riavvio...
Toolbox.FirmwareUpdater.Messages.ReadingDataflash=Leggendo dataflash...
Toolbox.FirmwareUpdater.Messages.WritingDataflash=Scrivendo dataflash...
Toolbox.FirmwareUpdater.Messages.RestartingDevice=Riavviando il dispositivo...
Toolbox.FirmwareUpdater.Messages.UploadingFirmware=Caricando il firmware...
Toolbox.FirmwareUpdater.Messages.DataflashReadAndSave=Dataflash letto con successo e salvato nel file.
Toolbox.FirmwareUpdater.Messages.DataflashWritten=Dataflash scritto con successo.
Toolbox.FirmwareUpdater.Messages.DataflashReseted=Dataflash resettato.

mirko93s, 2017-02-08:
I tested it a lot... Tell me if something is wrong

TBXin, 2017-02-09:
@mirko93s good job!

Everything is ok, except one little thing:

[image]

mirko93s, 2017-02-10:
Ok, corrected 😄 

// Shared
Toolbox.Shared.SecondString=secondo
Toolbox.Shared.SecondsString=secondi

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=Configurazione ArcticFox
Toolbox.MainWindow.MyEvicConfigurationButton=Configurazione myEvic
Toolbox.MainWindow.DeviceMonitorButton=Monitoraggio dispositivo
Toolbox.MainWindow.ScreenshooterButton=Istantanea Schermo
Toolbox.MainWindow.FirmwareUpdaterButton=Aggiornamento Firmware
Toolbox.MainWindow.AboutLink=Informazioni && Link

Toolbox.MainWindow.Tray.ShowToolbox=Mostra finestra NFE Toolbox
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Apri Configuratore ArcticFox con il dispositivo connesso.
Toolbox.MainWindow.Tray.TimeSync=Sincronizza orologio quando il dispositivo è collegato
Toolbox.MainWindow.Tray.StartWithWindows=Apri NFE Toolbox all'avvio di Windows
Toolbox.MainWindow.Tray.Exit=Esci

// ArcticFox Configuration Window
Toolbox.ArcticFoxConfiguration.Message.ConnectDevice=Connetti dispositivo con\n\nArcticFox\n[{0}]\n\nfirmware o più recente
Toolbox.ArcticFoxConfiguration.Message.DownloadingSettings=Download impostazioni...
Toolbox.ArcticFoxConfiguration.Message.OutdatedToolbox=NFE Toolbox non è aggiornato.\n\nPer proseguire, per favore scarica\n\nl'ultima versione disponibile.
Toolbox.ArcticFoxConfiguration.Message.NoCompatibleUSBDevice=Non è connessuno alcun dispositivo USB compatibile.\n\nPer proseguire, connettine uno.\n\nSe è già connesso, prova a scllegarlo e collegarlo di nuovo. Il cavo potrebbe essere lento.
Toolbox.ArcticFoxConfiguration.Message.UnableToReadData=Impossibile scaricare le impostazioni del dispositivo. Riconnetti il tuo dispositivo.

Toolbox.ArcticFoxConfiguration.Status.Device=Il dispositivo è
Toolbox.ArcticFoxConfiguration.Status.Connected=connesso
Toolbox.ArcticFoxConfiguration.Status.Disconnected=disconnesso
Toolbox.ArcticFoxConfiguration.Status.Ready=Pronto
Toolbox.ArcticFoxConfiguration.Status.OperationCompleted=Operazione completata

Toolbox.ArcticFoxConfiguration.ActiveProfileLabel=Attivo:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Batteria 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Batteria 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Batteria 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Batteria 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Modifica
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Modello Batteria:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=Controlla TCR:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.LSLLabel=Light Sleep Mode:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Limitatore Tiro:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=RCOBC:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Correzione Shunt:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=Ricarica USB:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=Nessun Risparmio:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.X32Label=X32:
Toolbox.ArcticFoxConfiguration.AdvancedTab=Avanzato
Toolbox.ArcticFoxConfiguration.BVOTab=BVO
Toolbox.ArcticFoxConfiguration.CircleScreenTab=Tema Cerchio
Toolbox.ArcticFoxConfiguration.ClassicScreenTab=Tema Classico
Toolbox.ArcticFoxConfiguration.ConfigurationButton=Configurazione▾
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.New=Nuovo
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.Open=Apri
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.SaveAs=Salva con nome
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=Connetti e Scarica configurazione
Toolbox.ArcticFoxConfiguration.ControlsTab=Controlli
Toolbox.ArcticFoxConfiguration.DownloadButton=Scarica impostazioni
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=Abilitato
Toolbox.ArcticFoxConfiguration.General.Controls.2ClicksLabel=2 Click:
Toolbox.ArcticFoxConfiguration.General.Controls.3ClicksLabel=3 Click:
Toolbox.ArcticFoxConfiguration.General.Controls.4ClicksLabel=4 Click:
Toolbox.ArcticFoxConfiguration.General.Controls.Power1WStepLabel=Potenza step 1W:
Toolbox.ArcticFoxConfiguration.General.Controls.SharedLabel=Condiviso:
Toolbox.ArcticFoxConfiguration.General.Controls.TCModeClicksLabel=Modalità TC Multi Click Pulsante Fire:
Toolbox.ArcticFoxConfiguration.General.Controls.UpDownButtonsLabel=Pulsanti Su/Giù:
Toolbox.ArcticFoxConfiguration.General.Controls.VWModeClicksLabel=Modalità VW Multi Click Pulsante Fire:
Toolbox.ArcticFoxConfiguration.General.Controls.WakeUpPlusMinusLabel=Risveglia con + / - :
Toolbox.ArcticFoxConfiguration.General.Layout.Line1Label=Linea 1:
[... 213 lines cut ...]
Toolbox.FirmwareUpdater.HardwareVersionLabel=Versione hardware:
Toolbox.FirmwareUpdater.BootModeLabel=Modalità di boot:
Toolbox.FirmwareUpdater.ProgressGroupLabel=Progresso:
Toolbox.FirmwareUpdater.OperationsGroupLabel=Operazioni:
Toolbox.FirmwareUpdater.CommonTab=Comune
Toolbox.FirmwareUpdater.UpdateButton=Aggiorna da file
Toolbox.FirmwareUpdater.DatalfashTab=Dataflash
Toolbox.FirmwareUpdater.ReadDataflashButton=Leggi dataflash
Toolbox.FirmwareUpdater.WriteDataflashButton=Scrivi dataflash
Toolbox.FirmwareUpdater.ResetDataFlashButton=Resetta dataflash
Toolbox.FirmwareUpdater.AdvancedTab=Avanzato
Toolbox.FirmwareUpdater.ChangeHWVerButton=Cambia versione HW
Toolbox.FirmwareUpdater.SwitchBootModeButton=Cambia modalità boot
Toolbox.FirmwareUpdater.CancelButton=Annulla

[... cut ...]
Toolbox.FirmwareUpdater.Messages.ReadingDataflash=Leggendo dataflash...
Toolbox.FirmwareUpdater.Messages.WritingDataflash=Scrivendo dataflash...
Toolbox.FirmwareUpdater.Messages.RestartingDevice=Riavviando il dispositivo...
Toolbox.FirmwareUpdater.Messages.UploadingFirmware=Caricando il firmware...
Toolbox.FirmwareUpdater.Messages.DataflashReadAndSave=Dataflash letto con successo e salvato nel file.
Toolbox.FirmwareUpdater.Messages.DataflashWritten=Dataflash scritto con successo.
Toolbox.FirmwareUpdater.Messages.DataflashReseted=Dataflash resettato.