TBXin/NFirmwareEditor #173: Adaptation of language packages

BigBadBlackWolf, 2017-02-11:
Since several questions were asked, why the myEvic button does not work in NFE Toolbox, the language packages should be adapted by a small change. Here is my suggestion for the English language pack:

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox Configuration
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic Configuration\n(not supported yet)

and for the German language pack:

BigBadBlackWolf, 2017-02-11:
// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox konfigurieren
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic konfigurieren\n(derzeit inaktiv)

All other language packages please also adjust.

BigBadBlackWolf, 2017-02-11:
Or we leave this button completely away, as long as myevic is not supported. If this is the case, it can be inserted again. Thus, no requests are made by users why this button does not work.

MoDuLah, 2017-02-22:
Greek:
// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=Ρυθμίσεις ArcticFox
Toolbox.MainWindow.MyEvicConfigurationButton=Ρυθμίσεις myEvic\n(Δεν υποστηρίζεται ακόμα)

closed 2017-05-08 by TBXin