TBXin/NFirmwareEditor #182: Pico Dual - Recognize Device Bug

westor7, 2017-02-20:
Hello,

I had just buy the Pico Dual and when i place the USB cable the program cannot recognize the device, i had try it with the default Eleaf program and it recognized so i assume it's program issue.

Check this video to reproduce it: https://goo.gl/XT2DlI

- Thanks!

TBXin, 2017-02-20:
@westor7 

Hello,

You should install ArcticFox firmware to get working ArcticFox configuration, i think this is obviously.

And Eleaf program is not configuration utility, it's firmware updater. You can press "Firmware Updater" button in the NToolbox and see that all are working fine. And one thing - you can install ArcticFox firmware only via NToolbox - Firmware Updater.

Best regards.

westor7, 2017-02-20:
Yes but the problem when i use Firmware Updater it says that:

https://goo.gl/s630RL

TBXin, 2017-02-20:
That is not problem, just missing correlation between PID (product identifier) and the real name.
The name of the device for convenience only, it is not affect anything.

> It shouldn't say Unknown Device yes?

I will fix this behavior later. Also need to add names for eVic Primo and some other devices.

TBXin, 2017-02-20:
You can try latest [nightly build](https://ci.appveyor.com/project/TBXin/nfirmwareeditor/build/artifacts).

closed 2017-05-08 by TBXin