TBXin/NFirmwareEditor #183: [Language Pack] NToolbox - Greek

MoDuLah, 2017-02-22:
// Shared
Toolbox.Shared.SecondString=δευτερόλεπτο
Toolbox.Shared.SecondsString=δευτερόλεπτα

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=Ρυθμίσεις ArcticFox
Toolbox.MainWindow.MyEvicConfigurationButton=Ρυθμίσεις myEvic
Toolbox.MainWindow.DeviceMonitorButton=Ελεγκτής Συσκευής
Toolbox.MainWindow.ScreenshooterButton=Εικονογράφος
Toolbox.MainWindow.FirmwareUpdaterButton=Αναβάθμιση Firmware
Toolbox.MainWindow.AboutLink=Σχετικά && Σύνδεσμοι

Toolbox.MainWindow.Tray.ShowToolbox=Εμφάνιση παραθύρου NFE Toolbox
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Άνοιγμα των Ρυθμίσεων του ArcticFox με τη σύνδεση συσκευής
Toolbox.MainWindow.Tray.TimeSync=Συγχρονισμός ώρας με τη σύνδεση συσκευής
Toolbox.MainWindow.Tray.StartWithWindows=Εκτέλεση του NFE Toolbox με την εκκίνηση των Windows
Toolbox.MainWindow.Tray.Exit=Έξοδος

// ArcticFox Configuration Window
Toolbox.ArcticFoxConfiguration.Message.ConnectDevice=Συνδέστε συσκευή με το \n\nArcticFox\n[{0}]\n\nfirmware ή νεώτερο
Toolbox.ArcticFoxConfiguration.Message.DownloadingSettings=Κατέβασμα ρυθμίσεων...
Toolbox.ArcticFoxConfiguration.Message.OutdatedToolbox=Η έκδοση του NFE Toolbox είναι ξεπερασμένη.\n\nΓια να συνεχίσετε, παρακαλώ κατεβάστε\n\nτην τελευταία διαθέσιμη έκδοση.
Toolbox.ArcticFoxConfiguration.Message.NoCompatibleUSBDevice=Δεν υπάρχουν συμβατές συσκευές USB που να είναι συνδεδεμένες.\n\nΓια να συνεχίσετε, παρακαλώ συνδέστε μία.\n\nΕάν η συσκευή είναι συνδεδεμένη, προσπαθήστε να την αποσυνδέσετε και να την συνδέσετε και πάλι. Το καλώδιο μπορεί να είναι χαλαρό.
Toolbox.ArcticFoxConfiguration.Message.UnableToReadData=Δεν είναι δυνατή η λήψη των ρυθμίσεων της συσκευής. Συνδέστε ξανά τη συσκευή σας.

Toolbox.ArcticFoxConfiguration.Status.Device=Η συσκευή έχει
Toolbox.ArcticFoxConfiguration.Status.Connected=συνδεθεί
Toolbox.ArcticFoxConfiguration.Status.Disconnected=αποσυνδεθεί
Toolbox.ArcticFoxConfiguration.Status.Ready=Έτοιμο
Toolbox.ArcticFoxConfiguration.Status.OperationCompleted=Η διαδικασία ολοκληρώθηκε

Toolbox.ArcticFoxConfiguration.ActiveProfileLabel=Ενεργό:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Μπαταρία 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Μπαταρία 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Μπαταρία 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Μπαταρία 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Επεξεργασία
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Μάρκα Μπαταρίας:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=Έλεγχος TCR:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.LSLLabel=ΛΕΥ (LSL):
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Μέγιστη Εισπνοή:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=ΕΜΚΑΜ (RCOBC):
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Ακρίβεια:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=Φόρτηση USB:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=Άυπνο USB:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.X32Label=X32:
Toolbox.ArcticFoxConfiguration.AdvancedTab=Για Προχωρημένους
Toolbox.ArcticFoxConfiguration.BVOTab=Ρυθμίσεις Μπαταριών
Toolbox.ArcticFoxConfiguration.CircleScreenTab=Κυκλική Οθόνη
Toolbox.ArcticFoxConfiguration.ClassicScreenTab=Κλασική Οθόνη
Toolbox.ArcticFoxConfiguration.ConfigurationButton=Διαμόρφωση▾
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.New=Δημιουργία
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.Open=Άνοιγμα
Toolbox.ArcticFoxConfiguration.ConfigurationMenu.SaveAs=Αποθήκευση Ώς
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=Συνδέστε τη συσκευή και κάντε λήψη των ρυθμίσεων
Toolbox.ArcticFoxConfiguration.ControlsTab=Πλήκτρα Ελέγχου
Toolbox.ArcticFoxConfiguration.DownloadButton=Λήψη Ρυθμίσεων
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=Ενεργό
Toolbox.ArcticFoxConfiguration.General.Controls.2ClicksLabel=2 Clicks:
Toolbox.ArcticFoxConfiguration.General.Controls.3ClicksLabel=3 Clicks:
Toolbox.ArcticFoxConfiguration.General.Controls.4ClicksLabel=4 Clicks:
Toolbox.ArcticFoxConfiguration.General.Controls.Power1WStepLabel=Δύναμη με βήμα 1W:
Toolbox.ArcticFoxConfiguration.General.Controls.SharedLabel=Κοινά:
Toolbox.ArcticFoxConfiguration.General.Controls.TCModeClicksLabel=TC Κατάσταση Fire Button Multi Clicks:
Toolbox.ArcticFoxConfiguration.General.Controls.UpDownButtonsLabel=Κουμπιά Πάνω/Κάτω:
Toolbox.ArcticFoxConfiguration.General.Controls.VWModeClicksLabel=VW Κατάσταση Fire Button Multi Clicks:
Toolbox.ArcticFoxConfiguration.General.Controls.WakeUpPlusMinusLabel=Αφύπνιση συσκευής με + / - :
Toolbox.ArcticFoxConfiguration.General.Layout.Line1Label=Γραμμή 1:
Toolbox.ArcticFoxConfiguration.General.Layout.Line2Label=Γραμμή 2:
Toolbox.ArcticFoxConfiguration.General.Layout.Line3Label=Γραμμή 3:
[... 211 lines cut ...]
Toolbox.FirmwareUpdater.HardwareVersionLabel=Έκδοση Υλικού:
Toolbox.FirmwareUpdater.BootModeLabel=Εκκίνηση:
Toolbox.FirmwareUpdater.ProgressGroupLabel=Πρόοδος:
Toolbox.FirmwareUpdater.OperationsGroupLabel=Λειτουργίες:
Toolbox.FirmwareUpdater.CommonTab=Αναβάθμιση
Toolbox.FirmwareUpdater.UpdateButton=Από αρχείο
Toolbox.FirmwareUpdater.DatalfashTab=Δεδομένα
Toolbox.FirmwareUpdater.ReadDataflashButton=Ανάγνωση Δεδομένων
Toolbox.FirmwareUpdater.WriteDataflashButton=Εγγραφή Δεδομένων
Toolbox.FirmwareUpdater.ResetDataFlashButton=Επαναφορά Δεδομένων
Toolbox.FirmwareUpdater.AdvancedTab=Προχωρημένες
Toolbox.FirmwareUpdater.ChangeHWVerButton=Αλλαγή έκδοσης Υλικού
Toolbox.FirmwareUpdater.SwitchBootModeButton=Αλλαγή τύπου εκκίνησης
Toolbox.FirmwareUpdater.CancelButton=Άκυρο

Toolbox.FirmwareUpdater.Messages.DeviceIsReady=Η συσκευή είναι έτοιμη.
Toolbox.FirmwareUpdater.Messages.WaitingForDevice=Αναμονή για συσκευή...
Toolbox.FirmwareUpdater.Messages.WaitingForDeviceAfterReset=Αναμονή για συσκευή μετά την επανεκκίνηση...
Toolbox.FirmwareUpdater.Messages.ReadingDataflash=Ανάγνωση μνήμης δεδομένων...
Toolbox.FirmwareUpdater.Messages.WritingDataflash=Εγγραφή μνήμης δεδομένων...
Toolbox.FirmwareUpdater.Messages.RestartingDevice=Επανεκκίνηση Συσκευής...
Toolbox.FirmwareUpdater.Messages.UploadingFirmware=Μεταφόρτωση Λογισμικού...
Toolbox.FirmwareUpdater.Messages.DataflashReadAndSave=Η μνήμη δεδομένων αναγνώστικε και γράφτικε επιτυχώς στο αρχέιο.
Toolbox.FirmwareUpdater.Messages.DataflashWritten=Επιτυχής εγγραφή μνήμης δεδομένων.
Toolbox.FirmwareUpdater.Messages.DataflashReseted=Η μνήμη δεδομένων επαναφέρθηκε.

westor7, 2017-02-24:
You can replace "Εισπνοή" with "Τζούρα" to be more understandable.

MoDuLah, 2017-02-25:
@westor7 
A) I could see that coming about εισπνοή/τζούρα but the word "τζούρα" is the part of the smoking/vaping slang.
B) Yeah ok clicks can be in greek.
C) I let it as TCR Error because when it happens you get that on device and it shows TCR Error. Unable to translate on device.
D) Sure.
E) Toolbox.ArcticFoxConfiguration.InfoLines.NonDominant=Μη Κυρίαρχο (Ισχ. / Θερμ.) maybe is even better.

westor7, 2017-02-25:
@MoDuLah 

Yes the "Τζούρα" it actually refers to smoking first but personally i prefer it than "Εισπνοή" because at any vaping community or any other vaping resource, that word doesn't exist, as "Τζούρα" that it still some vapers use it as the head core word that they want to specify the puff, anyway that one isn't something big, so good work mate.

GnFlala, 2017-02-27:
[GR.lpack.txt](https://github.com/TBXin/NFirmwareEditor/files/804851/GR.lpack.txt)
Based on @MoDuLah translation with a lot of fixes and changes. 

Good job @MoDuLah. Thanks man

westor7, 2017-03-01:
Because some translations are too big in result the meaning being lost, i suggest to modify them. (tested and are working better)

Toolbox.ArcticFoxConfiguration.DownloadButton=Λήψη Ρυθμ.
Toolbox.ArcticFoxConfiguration.UploadButton=Εφαρμογή Ρυθμ.
Toolbox.ArcticFoxConfiguration.ResetButton=Επαναφορά Ρυθμ.

Also you can change all "Volts" to "Βόλτ" .

GnFlala, 2017-03-03:
agreed on the 3 lines.

But as far Volts.... I am an electrician. The global sign for Volts is V. Νοt Β. I have never seen anywhere even on greek service manuals, electricity diagrams, or any kind of diagrams the greek work βολτ used. I would point to the global used one "V" and Volts.

westor7, 2017-03-03:
@GnFlala Yeah ok about the Volts look here (https://translate.google.com/#auto/el/volts) it also translated in Greek :) i agree about the sign to be V as W for watt but you can translate also the word to greek for safe because it is valid.

MoDuLah, 2017-06-16:
Update coming soon. The GR Flag is here though
[image]

TBXin, 2017-06-16:
https://nfeteam.org/forum/forums/nfe-tools-localizations/

MoDuLah, 2017-06-16:
I will upload it there as soon as it's done. ;) Cheers TBXin Keep up the good work.