TBXin/NFirmwareEditor #189: [ADD FEATURE] 2nd instance

Pilou59, 2017-02-26:
Hi all,

I'm testing back TC with my new Reuleaux RX300. So I'm looking for the good settings with 316L coil. I'd like to use PI reg. I'm testing different values for Watt, P and I value, preheat (curve and others)
With the device monitor, I can have a full monotoring of my resistance value, temp, amp...

So, it would be so useful to have the possibilty to open a second instance of nfe toolbox to have arcticfox config beside device monitor.

TBXin, 2017-02-27:
Hello,

At this point i am not doing this, because it's a bit complicated to synchronize input and output streams.
I will think about what i can do with this, but I can't promise anything.

Pilou59, 2017-02-28:
And what about a "all in one" program? It could be simplier? Also a lot of coding hours I know... But it would be so great ;-)

TBXin, 2017-02-28:
@Pilou59 now like it is, all in one program or you meant something else?

smok-e, 2017-03-01:
I think an "all in one program" solution would be better: when you could just have the device monitor display floating, while you could change the settings of the device in the configuration window. 
Or in other words: that you could open both from the NFEtoolbox window at the same time.

closed 2017-06-16 by TBXin