TBXin/NFirmwareEditor #190: Tabulation Order in Curve editor

ilyxa, 2017-02-27:
**Issue Descriptions:**
Wrong tabbing navigation in any curve editor
**Steps to Reproduce:**
Open **Arcticfox Configuration** -> **Advanced** -> **Power Curves** -> **any**
Press Tab -> Cancel (selected) -> <Tab> -> Save (selected) -> .. -> Import -> .. -> Export -> cycle.

More convenient behavior: <Tab> - Profile Name -> <Tab> -> F (1st field) -> <Tab> -> Factor (1st field) -> and so on.

Also it would be great to add shortcuts to the buttons Import / Export / Save / Cancel on all interface (add shortcuts, please).

TBXin, 2017-02-27:
Hello,

This is a known issue, but a little lazy to fix the tab order ^^

TBXin, 2017-02-28:
Thanks for commit ^^

closed 2017-02-28 by TBXin