TBXin/NFirmwareEditor #192: High CPU usage with Device Monitor

snigel, 2017-03-05:
When I use the device monitor the cpu usage creeps up until it consumes one core. From then it only becomes slower over time until it has to be restarted. It seems like increasing the X axis to 60 seconds make this happen faster. 15 minutes of run time is probably enough.

I've reproduced this several times on two different machines, Win 7 64bit and Win 10 64bit, together with an Eleaf iPower 80w.

TBXin, 2017-03-05:
Hi,

What is your processor?

snigel, 2017-03-06:
Win 7 64bit: Intel i7 6820HQ
Win 10 64bit: Intel i5 860

snigel, 2017-03-08:
I checked more. Resized the window and set X scale to 60 seconds.
Took me about 5 minutes to spike one core.

Windows can be a bit tricky with showing that a core goes full, manually setting affinity for a process shows this more clearly.

GnFlala, 2017-03-15:
I have the exact same problem. After a while cpu usage is high (goes up to 25 % when affinity is on all cores and 100% if it is on one core.). Device monitor is laggy after that happens.

closed 2017-06-16 by TBXin