TBXin/NFirmwareEditor #195: NFirmware editor for RX300?

Echo113, 2017-03-16:
Ive copied the rx300 5.04.xml file posted on the other page to the Directories but im still getting unable to load firmware. No one definition is not appropriate for the selected firmware file.
[image]

maelstrom2001, 2017-03-16:
Yep, firmware was updated, use attached definition for new binary.

[RX300 5.04 (13.01.2017).zip](https://github.com/TBXin/NFirmwareEditor/files/848882/RX300.5.04.13.01.2017.zip)

Echo113, 2017-03-16:
That worked! Thanks! Do you happen to know how to invert the white squares around the ohms, Volts, etc?
[image]

maelstrom2001, 2017-03-16:
> how to invert the white squares

By modifying firmware. Glyphs are drawn inverted at that places, need to replace calls of drawing procedures to regular, without inversion.

Echo113, 2017-03-17:
Yea I know that much but the doesn't seem to be a bmp to invert. I've tried inverting literally everything. It must be coded to be white.

closed 2017-05-08 by TBXin