TBXin/NFirmwareEditor #200: [Question] Decompiling firmware

IIsSchulstad, 2017-03-24:
I'm sorry to bother you, but may i ask how you managed to read the firmware files?
My vape is not supported by your tool and im not asking you to make it compatible, im just asking this for my own interest.
I can open the firmware in a hex-editing tool and change the strings, no problem, but how are you able to edit the rest of the code, eg. the bitmaps? I mean the firmwares are compiled, right?

I hope this question din't annoy you too much

TBXin, 2017-03-24:
Hello,

If we are talking about Joyetech/Wismec/Eleaf firmwares - yes, this is compiled from sources binary files.
If you have device from another manufacture - i don't know, because you are not specify what device you have.

NFirmwareEditor is an open-source project and you can look in the sources and look how NFE is read images ^^

IIsSchulstad, 2017-03-24:
I have a Alien mod from Smok, I will have a look a look at your code.
Thanks for the reply :)

TBXin, 2017-03-25:
@IIsSchulstad oh, algorithms that I use to open Joyetech firmwares is not suitable for Smok firmwares.

NFirmwareEditor is parsing compiled binary files and looking for resources tables, which are presents because of specifics of the Joyetech firmwares.

I don't know which algorithms Smok uses to store resources. This is research work and required more time to investigate. At this moment i am not interested in this.

closed 2017-03-25 by IIsSchulstad