TBXin/NFirmwareEditor #208: NFirmwareEditor for Predator 228?

ETPNV13, 2017-04-08:
**Issue Descriptions:**
Just wondering when/if there is going to be an update to edit the Predator 228 firmware which I think is pretty similar to the rx300 firmware...wouldn't even ask but there has not been an update here in two months which is kind of weird/worrying to me since there is usually one like once a month
**Steps to Reproduce:**

TBXin, 2017-04-08:
Hello,

Unzip file and put it in the "Definitions" directory, which is located near NFirmwareEditor.exe

[Predator 228 5.04.zip](https://github.com/TBXin/NFirmwareEditor/files/907509/Predator.228.5.04.zip)

ETPNV13, 2017-04-09:
Worked perfectly, thank you much!

ETPNV13, 2017-04-16:
Just another quick question, is there a way to get rid of the inverted portions and the arch over the wattage? I used to just delete the 4 circles on older firmware on other devices but there are no rectangular boxes and the arch is nowhere to be found either...If there is an easy way to do this that would be great but if not it is fine because as soon as arctic fox is compatible I am switching to that firmware anyways

zxr750, 2017-05-06:
he dont work my mod...
[http://image.prntscr.com/image/aaf2eca6e46c4fefa761d9c218169f8d.png](url)

zxr750, 2017-05-06:
[http://prnt.sc/f4rcne](url)

TBXin, 2017-05-06:
This is an issue about editing OFW, not installing ArcticFox.

closed 2017-05-08 by TBXin