TBXin/NFirmwareEditor #210: cks icon 200w firmware

rizanfauzan, 2017-04-22:
can you please add defenitions for cks icon 200w firmware
i need it because the firmware that based on vo chipset missing some string la coma, battery indicator, number 9, and make my mod display error
please
pardon my english

TBXin, 2017-04-22:
Sorry but no.

This software is designed to open & edit only Joyetech based PCB's firmwares.
I don't know anything about VO chipset and their firmware.

rizanfauzan, 2017-04-22:
thats a shame
thank you for replying though
i really appreciate it

TBXin, 2017-04-22:
shame? Oo

rizanfauzan, 2017-04-22:
sorry for my english
i mean it as im unlucky or something
i probably use the word wrong T_T

TBXin, 2017-04-22:
Okay, bro :) 
Thanks for the explanation!

maelstrom2001, 2017-04-22:
@rizanfauzan Look at title: "Firmware editor for vape devices from Joyetech, Wismec, Eleaf"
Do you think it's very simple and enough to add a single file to support a **completely different** device?

rizanfauzan, 2017-04-22:
@maelstrom2001 sorry, because the title says "and so on" that i think its almost for most device
for the inconvenience im sorry to bother
im not trying to make users here mad
@TBXin sorry and thanks for understanding

closed 2017-05-08 by TBXin