TBXin/NFirmwareEditor #216: [Language Pack] NToolbox - Chinese

vojaly, 2017-05-11:
[CN.lpack.zip](https://github.com/TBXin/NFirmwareEditor/files/992319/CN.lpack.zip)

// Shared
// Edit by Vojaly

Toolbox.ArcticFoxConfiguration.AdvancedTab=高级
Toolbox.ArcticFoxConfiguration.ConfigurationButton=配置▾
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=连接设备并下载配置信息
Toolbox.ArcticFoxConfiguration.ControlsTab=控制
Toolbox.ArcticFoxConfiguration.DownloadButton=下载设置
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=开启
Toolbox.ArcticFoxConfiguration.FWVerLabel=固件:
Toolbox.ArcticFoxConfiguration.HWVerLabel=硬件:
Toolbox.ArcticFoxConfiguration.NewConfigurationLink=创建新的配置
Toolbox.ArcticFoxConfiguration.OpenConfigurationLink=打开现有的配置
Toolbox.ArcticFoxConfiguration.ProfilesTab=雾化器配置
Toolbox.ArcticFoxConfiguration.ResetButton=复位设置
Toolbox.ArcticFoxConfiguration.ScreenTab=屏幕
Toolbox.ArcticFoxConfiguration.SecondsLabel=秒
Toolbox.ArcticFoxConfiguration.SmartTab=智能
Toolbox.ArcticFoxConfiguration.StatsTab=统计
Toolbox.ArcticFoxConfiguration.UploadButton=上传设置
Toolbox.ArcticFoxConfiguration.VoltsLabel=V
Toolbox.ArcticFoxConfiguration.Advanced.BVOTab=电压数值修正
Toolbox.ArcticFoxConfiguration.Advanced.MaterialsTab=热丝曲线
Toolbox.ArcticFoxConfiguration.Advanced.PowerCurvesTab=功率曲线
Toolbox.ArcticFoxConfiguration.Advanced.SettingsTab=设置
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=电池 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=电池 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=电池 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=电池 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=编辑
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=电池型号:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=检查温控(TCR):
Toolbox.ArcticFoxConfiguration.Advanced.Settings.DeepSleepBehaviorLabel=深度睡眠模式:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=单次最长点火:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=充电时统计清零:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RTCModeLabel=RTC 模式:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=并联校正:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.TooltipLabel=将鼠标移动到该复选框或组合框或标签来显示帮助提示。
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=USB充电:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=USB连接时不睡眠:
Toolbox.ArcticFoxConfiguration.BatteryModel.Generic=通用电池
Toolbox.ArcticFoxConfiguration.ChargeScreen.Classic=经典屏幕
Toolbox.ArcticFoxConfiguration.ChargeScreen.Extended=扩展屏幕
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.AnalogClock=模拟时钟
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.DigitalClock=数字时钟
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.Logo=Logo
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.None=空
Toolbox.ArcticFoxConfiguration.ClickActions.DeviceLockUnlock=设备上锁 / 解锁
Toolbox.ArcticFoxConfiguration.ClickActions.DeviceOnOff=设备 开机 / 关机
Toolbox.ArcticFoxConfiguration.ClickActions.Edit=编辑
Toolbox.ArcticFoxConfiguration.ClickActions.Edit Profile=编辑配置
Toolbox.ArcticFoxConfiguration.ClickActions.InfoScreen=显示信息
Toolbox.ArcticFoxConfiguration.ClickActions.KeyLockUnlock=键盘锁 / 解锁
Toolbox.ArcticFoxConfiguration.ClickActions.LLSOnOf=LSL On / Off
Toolbox.ArcticFoxConfiguration.ClickActions.MainMenu=主菜单
Toolbox.ArcticFoxConfiguration.ClickActions.MenuBack=返回
Toolbox.ArcticFoxConfiguration.ClickActions.MenuExit=退出
Toolbox.ArcticFoxConfiguration.ClickActions.None=无行为
Toolbox.ArcticFoxConfiguration.ClickActions.PowerBank=充电宝功能
Toolbox.ArcticFoxConfiguration.ClickActions.PreheatMenu=预热菜单
Toolbox.ArcticFoxConfiguration.ClickActions.ResestSavedResistance=重置已保存电阻
Toolbox.ArcticFoxConfiguration.ClickActions.ResetCounters=重置计数器
Toolbox.ArcticFoxConfiguration.ClickActions.ResistanceLockUnlock=电阻 锁止 / 解锁
Toolbox.ArcticFoxConfiguration.ClickActions.SelectProfile=选择配置
Toolbox.ArcticFoxConfiguration.ClickActions.ShowClock=显示时钟
Toolbox.ArcticFoxConfiguration.ClickActions.SmartOnOff=智能模式 开启 / 关闭
Toolbox.ArcticFoxConfiguration.ClickActions.StealthOnOff=屏幕节能 开启 / 关闭
Toolbox.ArcticFoxConfiguration.ClickActions.TDom=TDom
[... 240 lines cut ...]
Toolbox.MainWindow.ScreenshooterButton=截屏助手
Toolbox.MainWindow.TaglineLabel=你将能获得无极限的体验
Toolbox.MainWindow.TitleLabel=欢迎使用NToolbox
Toolbox.MainWindow.ArcticFoxConfigurationButton.Tooltip=配置你的设备，体验强大的北极狐固件功能。
Toolbox.MainWindow.DeviceMonitorButton.Tooltip=读取所有传感器的数据
Toolbox.MainWindow.FirmwareUpdaterButton.Tooltip=安装新固件或升级现有的固件
Toolbox.MainWindow.ScreenshooterButton.Tooltip=分享你的设备截图
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=连接你的设备并配置北极狐设置。
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceDisconnected=您的设备已断开，请关闭配置软件。
Toolbox.MainWindow.Tray.Exit=退出
Toolbox.MainWindow.Tray.ShowToolbox=显示NFE Toolbox窗口。
Toolbox.MainWindow.Tray.StartWithWindows=当系统启动时，自动运行NFE软件。
Toolbox.MainWindow.Tray.TimeSync=连接时，自动同步设备。
Toolbox.Screenshooter.CancelButton=取消
Toolbox.Screenshooter.ControlsGroupLabel=控制:
Toolbox.Screenshooter.PixelSizeLabel=像素尺寸计算:
Toolbox.Screenshooter.SaveScreenshotButton=保存截图
Toolbox.Screenshooter.ScreenGroupLabel=屏幕:
Toolbox.Screenshooter.StartBroadcastButton=捕获屏幕
Toolbox.Screenshooter.StopBroadcastButton=停止捕获
Toolbox.Screenshooter.TakeBeforeSaveCheckbox=开启
Toolbox.Screenshooter.TakeBeforeSaveLabel=截屏前保存:
Toolbox.Screenshooter.TakeScreenshotButton=获取屏幕图片
Toolbox.Shared.SecondsString=秒
Toolbox.Shared.SecondString=秒（点击后，设备会自动点火测试。）

TBXin, 2017-05-11:
Thanks. Integrated.

vojaly, 2017-05-11:
Thanks for you

vojaly, 2017-05-11:
Subsequent versions of Chinese. Can contact me in advance, I would like to help. EMAIL:1471229@qq.com