TBXin/NFirmwareEditor #218: [Language Pack] NToolbox - Turkish

utkucanc, 2017-05-15:
http://s4.dosya.tc/server4/v29k7l/ESF_NToolbox_Turkce_Yama_.rar.html
//edit by utkucanc and ibolister
//ESF Turkish E-Cigarette Public Forum
Toolbox.ArcticFoxConfiguration.AdvancedTab=Gelişmiş
Toolbox.ArcticFoxConfiguration.ConfigurationButton=Konfigürasyonlar▾
Toolbox.ArcticFoxConfiguration.ConnectAndDownloadLink=Konfigürasyonu Bağla ve İndir
Toolbox.ArcticFoxConfiguration.ControlsTab=Kontroller
Toolbox.ArcticFoxConfiguration.DownloadButton=Ayarları İndir
Toolbox.ArcticFoxConfiguration.EnabledCheckBox=Etkin
Toolbox.ArcticFoxConfiguration.FWVerLabel=FW Ver:
Toolbox.ArcticFoxConfiguration.HWVerLabel=HW Ver:
Toolbox.ArcticFoxConfiguration.NewConfigurationLink=Yeni yapılandırma oluştur
Toolbox.ArcticFoxConfiguration.OpenConfigurationLink=Mevcut yapılandırmayı aç
Toolbox.ArcticFoxConfiguration.ProfilesTab=Profiller
Toolbox.ArcticFoxConfiguration.ResetButton=Ayarları Sıfırla
Toolbox.ArcticFoxConfiguration.ScreenTab=Ekran
Toolbox.ArcticFoxConfiguration.SecondsLabel=s
Toolbox.ArcticFoxConfiguration.SmartTab=Akıllı
Toolbox.ArcticFoxConfiguration.StatsTab=İstatislikler
Toolbox.ArcticFoxConfiguration.UploadButton=Makinayı Güncelle
Toolbox.ArcticFoxConfiguration.VoltsLabel=v
Toolbox.ArcticFoxConfiguration.Advanced.BVOTab=BVO
Toolbox.ArcticFoxConfiguration.Advanced.MaterialsTab=Materyaller
Toolbox.ArcticFoxConfiguration.Advanced.PowerCurvesTab=Güç Eğrileri
Toolbox.ArcticFoxConfiguration.Advanced.SettingsTab=Ayarlar
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery1Label=Bateri 1:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery2Label=Bateri 2:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery3Label=Bateri 3:
Toolbox.ArcticFoxConfiguration.Advanced.BVO.Battery4Label=Bateri 4:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryEditButton=Düzenle
Toolbox.ArcticFoxConfiguration.Advanced.Settings.BatteryModelLabel=Bateri Modeli:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.CheckTCRLabel=TCR'yi kontrol et:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.DeepSleepBehaviorLabel=Derin Uyku:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.PuffCutOffLabel=Puff Kesme-Kapatma:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RCOBCLabel=RCOBC:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.RTCModeLabel=RTC Modu:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.ShuntCorrectionLabel=Shunt Correction:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.TooltipLabel=Araç ipucunu göstermek için fareyi onay kutusuna etikete taşıyın.
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBChargeLabel=USB Şarj:
Toolbox.ArcticFoxConfiguration.Advanced.Settings.USBNoSleepLabel=USB Uyku Modu:
Toolbox.ArcticFoxConfiguration.BatteryModel.Generic=Genel Pil
Toolbox.ArcticFoxConfiguration.ChargeScreen.Classic=Klasik
Toolbox.ArcticFoxConfiguration.ChargeScreen.Extended=Genişletilmiş
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.AnalogClock=Saat A
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.DigitalClock=Saat D
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.Logo=Logo
Toolbox.ArcticFoxConfiguration.ChargeScreenExtra.None=Yok
Toolbox.ArcticFoxConfiguration.ClickActions.DeviceLockUnlock=Cihaz Kilidi / Kilidi Açma
Toolbox.ArcticFoxConfiguration.ClickActions.DeviceOnOff=Cihaz Açık / Kapalı
Toolbox.ArcticFoxConfiguration.ClickActions.Edit=Düzenle
Toolbox.ArcticFoxConfiguration.ClickActions.Edit Profile=Profili Düzenle
Toolbox.ArcticFoxConfiguration.ClickActions.InfoScreen=Ekran Hakkında
Toolbox.ArcticFoxConfiguration.ClickActions.KeyLockUnlock=Kilidi Açma / Kapatma
Toolbox.ArcticFoxConfiguration.ClickActions.LLSOnOf=LSL Açık / Kapalı
Toolbox.ArcticFoxConfiguration.ClickActions.MainMenu=Ana Menü
Toolbox.ArcticFoxConfiguration.ClickActions.MenuBack=Geri
Toolbox.ArcticFoxConfiguration.ClickActions.MenuExit=Çıkış
Toolbox.ArcticFoxConfiguration.ClickActions.None=Eylem Yok
Toolbox.ArcticFoxConfiguration.ClickActions.PowerBank=Powerbank
Toolbox.ArcticFoxConfiguration.ClickActions.PreheatMenu=Isıtma Menüsü
Toolbox.ArcticFoxConfiguration.ClickActions.ResestSavedResistance=Kayıtlı Direnmeyi Sıfırla
Toolbox.ArcticFoxConfiguration.ClickActions.ResetCounters=Sayaçları Sıfırla
Toolbox.ArcticFoxConfiguration.ClickActions.ResistanceLockUnlock=Direnç Kilitleme / Kilidini Açma
Toolbox.ArcticFoxConfiguration.ClickActions.SelectProfile=Profil Seç
Toolbox.ArcticFoxConfiguration.ClickActions.ShowClock=Saati Göster
Toolbox.ArcticFoxConfiguration.ClickActions.SmartOnOff=Akıllı Mod Açık / Kapalı
Toolbox.ArcticFoxConfiguration.ClickActions.StealthOnOff=Gizlilik Modu Açık / Kapalı
Toolbox.ArcticFoxConfiguration.ClickActions.TDom=TDom
Toolbox.ArcticFoxConfiguration.ClockType.Analog=Analog
Toolbox.ArcticFoxConfiguration.ClockType.Digital=Dijital
[... 239 lines cut ...]
Toolbox.MainWindow.ScreenshooterButton=Ekran Görüntüsü
Toolbox.MainWindow.TaglineLabel=Türkçe arayüz UtkuCanC ve ibolister hazırlamıştır.(ESF)
Toolbox.MainWindow.TitleLabel=NToolbox'a hoş geldiniz.
Toolbox.MainWindow.ArcticFoxConfigurationButton.Tooltip=Cihazınızı ArcticFox firmware ile yapılandırın.
Toolbox.MainWindow.DeviceMonitorButton.Tooltip=Tüm sensör okumaları üzerinde tam kontrole sahip olun.
Toolbox.MainWindow.FirmwareUpdaterButton.Tooltip=Yeni firmware yükleyin yada mevcut olanı yükseltin.
Toolbox.MainWindow.ScreenshooterButton.Tooltip=Cihazın ekran görüntüsünü paylaş.
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceConnected=Cihaz bağlantısı sağlandı, ArcticFox Yapılandırılması açıldı.
Toolbox.MainWindow.Tray.ArcticFoxConfigurationWhenDeviceDisconnected=Cihaz bağlantısı kesildi, ArcticFox yapılandırması kapatıldı.
Toolbox.MainWindow.Tray.Exit=Çıkış
Toolbox.MainWindow.Tray.ShowToolbox=NFE Toolbox'ı ekranda görüntüle
Toolbox.MainWindow.Tray.StartWithWindows=NFE Toolbox bilgisayar açılışında açılsın.
Toolbox.MainWindow.Tray.TimeSync=Cihaz bağlıyken zamanı senkronize edin.
Toolbox.Screenshooter.CancelButton=İptal
Toolbox.Screenshooter.ControlsGroupLabel=Kontroller:
Toolbox.Screenshooter.PixelSizeLabel=Görüntü büyüklüğü çarpanı:
Toolbox.Screenshooter.SaveScreenshotButton=Save screenshot
Toolbox.Screenshooter.ScreenGroupLabel=Ekran:
Toolbox.Screenshooter.StartBroadcastButton=Kaydı Başlat
Toolbox.Screenshooter.StopBroadcastButton=Kaydı Durdur
Toolbox.Screenshooter.TakeBeforeSaveCheckbox=Aktif
Toolbox.Screenshooter.TakeBeforeSaveLabel=Şurdan önce kaydet:
Toolbox.Screenshooter.TakeScreenshotButton=Ekran Görüntüsü Al
Toolbox.Shared.SecondsString=Saniye
Toolbox.Shared.SecondString=Saniye

TBXin, 2017-05-16:
Thanks. Integrated!

utkucanc, 2017-06-02:
@TBXin 
291 - Toolbox.FirmwareUpdater.BootLabel=Boot
to
291- Toolbox.FirmwareUpdater.BootLabel=Ön Yükleme
:+1:

TBXin, 2017-06-02:
@utkucanc 
https://nfeteam.org/forum/forums/nfe-tools-localizations/

utkucanc, 2017-06-06:
@TBXin i create :+1: