TBXin/NFirmwareEditor #220: Import DNA Batterieprofiles

int3ks, 2017-05-18:
public BatteryProfile ImportCsv(FileStream file)
        {
            
            using (var sr = new StreamReader(file)) {
                var data = sr.ReadToEnd();
                var lines = data.Split(new[] { '\n', '\r' }, StringSplitOptions.RemoveEmptyEntries).ToList();
                lines = lines.Skip(1).ToList();
                var points = new List<BatteryProfilePoint>();
                foreach (var line in lines) {
                    var parts = line.Split(',');
                    points.Add(new BatteryProfilePoint {
                        Percent = (byte)Math.Round(double.Parse(parts[0], CultureInfo.InvariantCulture)),
                        Voltage = decimal.Parse(parts[1], CultureInfo.InvariantCulture) });
                }
                return new BatteryProfile { Cutoff = 2.8m, Points = points.ToArray() };
            }
        }

would be extremly helpfull...

TBXin, 2017-05-18:
And now you should handle cases when point count is not 11. 
DNA's discharge profile can contains any number of points, ArcticFox's discharge profile - not.

int3ks, 2017-05-18:
ok. i will modify this to read only 11 samples of data....

TBXin, 2017-05-18:
This will not work properly. This task is not so easy as you think.

int3ks, 2017-05-18:
your right. its not the easy. one question. where come this limitation from? chipset,firmware or the editor?

int3ks, 2017-05-18:
ok. i downloaded a few battery profiles from various sites and ALL of them have 11 values.
i know escribe can load any number of points but this would be better then nothing.
the source i modified brings up an errormessage if more or less points then 11  are inside of the csv file. 
this is absolute ok. and better then nothing...

in the function "importbutton_click" i also changed this ->
BatteryProfile data;
                    if (filePath.ToLower().EndsWith(".csv"))
                    {
                        data = ImportCsv(fs);
                    }
                    else
                    {
                        data = Serializer.Read<BatteryProfile>(fs);
                    }

and the filter for the openfiledialog has to be changed ;o)

smok-e, 2017-05-18:
just one question, when you have a profile with more than 11 points,
how would one decide which points to choose for the NFE profile?
I think it is not really working if you only give an error message.
It needs to be clear for the user, as there are many who don't understand it totally,
and then this will cause a lot of confusion.
I think it would be better to have a way outside of NFE toolbox to convert DNA profiles to NFE profiles.
If you can progamm something for this that would be great.
But still how will you convert a curve with more or less than 11 points correctly?
And how will you interpolate and convert the DNA values to the NFE values?

int3ks, 2017-05-18:
i understand that it is really complicatated to interpolate these datapoints. BUT ALL csv file downloaded has the required 11 points. in 99% there is no problem. so in this 1% we should only show an error message that csv files with 11 points are required.
this would make things easier for MOST user ;)
if you get an error message you can edit it in notepad and import it again,

int3ks, 2017-05-18:
btw.... i really really like the arctic fox firmware !!! it has all the features that evolv try to sell with the new dna75c chipset. @ the moment i use my cuboid with artic fox and i like it! my dna mods are waiting in the bookshelf... :D

i like it. big thx

smok-e, 2017-05-18:
I still don't understand how you will convert a DNA csv file to na valid NFE file.
I could upload some DNA curves where I put in only 2 points or I can make a csv file with 10 points from 4,20 volts to 4.10 volts and then 5 points from 3.50 volts to 3.20 volts. Or I can save this curve 
with just 11 points, but how will you convert it to a usefull NFE curve?

int3ks, 2017-05-19:
the source code i posted does this allready. i tested this with discharge profiles from known dna websites and it works with every single file!
i.e.
http://dna200.de/kurven_akku/
http://dna200-themes.de/dna75/

the problem that dna profiles can have any numbers of datapoints is only a theoretically problem. in fact the most of them have 11 points.

i can post a build of nfe if you like.

have fun
inteks

int3ks, 2017-05-19:
[NToolbox.dna.batt.profile.import.zip](https://github.com/TBXin/NFirmwareEditor/files/1013267/NToolbox.dna.batt.profile.import.zip)

closed 2017-06-16 by TBXin